attify/firmware-analysis-toolkit #1: Not working for me...

pabb85, 2016-11-08:
Dear Sir,

So I have the following output;
---begins---
root@kali:~/tools/firmadyne# ./fat.py 

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : Dlink_firmware.bin
Enter the brand of the firmware : Dlink
Dlink_firmware.bin
Now going to extract the firmware. Hold on..
/root/tools/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "Dlink_firmware.bin" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /root/tools/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
mke2fs 1.43.3 (04-Sep-2016)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu-system-mipsel: terminating on signal 2 from pid 1925
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Done!

Running the firmware finally : 
---ends---

But nothing is coming up, demonstrated like so;
---begins---
root@kali:~# nmap 192.168.0.1

Starting Nmap 7.31 ( https://nmap.org ) at 2016-11-08 20:11 GMT
Nmap scan report for 192.168.0.1
Host is up (0.00045s latency).
All 1000 scanned ports on 192.168.0.1 are filtered
MAC Address: 00:DE:FA:19:C0:02 (Unknown)

Nmap done: 1 IP address (1 host up) scanned in 34.69 seconds
---ends---

Any tips on where to start debugging this?

Cheers,
P.

pabb85, 2016-11-08:
Also this, if it is useful because the fat.py script refers to an interface not on my system;

---begins---
...
tap1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.2  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::c3a:e0ff:fe28:e790  prefixlen 64  scopeid 0x20<link>
        ether 0e:3a:e0:28:e7:90  txqueuelen 1000  (Ethernet)
        RX packets 98  bytes 24140 (23.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2070  bytes 119700 (116.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@kali:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 tap1
192.168.0.1     192.168.0.1     255.255.255.255 UGH   0      0        0 tap1
192.168.56.0    0.0.0.0         255.255.255.0   U     100    0        0 eth0
---ends---

adi0x90, 2016-11-09:
@pabb85 : Can you post the OS (Kali or Ubuntu) and the version?

pabb85, 2016-11-09:
Kali latest rolling version 64 bit

arunmagesh, 2016-11-09:
Hello @pabb85, 

Based on your log above everything looks good. 
Few things that you can try are
1.Have you tried to open this IP in a web-browser? 
2. It is a Firmware simulator in-order to debug and tinker with the firmware you can use FMK to extract the firmware to modify it and repackage and use FAT to check the result. 

If you have attended Attify's training. You can try the "Backdoor a firmware" exercise with FAT.

Cheers,
Arun

adi0x90, 2016-11-09:
@pabb85 : Also, can you attach both the logs located in `firmadyne/scratch/1/` folder to help us debug what might have went wrong?

pabb85, 2016-11-09:
Hi Arun,

I tried in the web browser first but no response - nmap was simply for a nicer way to show to you here that the port is not responding without a screenshot.

I am currently following Aditya's training and so it's currently a blocker on that, boo :(

Three things seem strange to me currently, the two lines;
"Please check the makeImage function" - which by tracing the python looks like a function has failed an caught an exception..... also,
"qemu-system-mipsel: terminating on signal 2 from pid 1925" process being killed doesn't seem okay?

The third thing is the interface showing on the script as br0 whilst the system interface is tap0 - maybe the br0 refers to the name of the interface on the emulated system or there's a hardcoded interface name causing an issue?  

I'll take another look in the python script and see if I can trace it through can you let me know if you have any updates please?

Best regards,
Paul

adi0x90, 2016-11-09:
@pabb85 : The qemu termination and the br0 looks okay and normal behaviour. 

Two things : 

1. Can you post the iptables rule of the Kali machine? 
2. Run the below commands and let us know if it works : 

```
./reset.sh
./sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "Dlink_firmware.bin" images
./scripts/getArch.sh ./images/1.tar.gz
./scripts/tar2db.py -i 1 -f ./images/1.tar.gz
sudo ./scripts/makeImage.sh 1
./scripts/inferNetwork.sh 1
./scratch/1/run.sh
```

pabb85, 2016-11-09:
[FAT_logs_091116.zip](https://github.com/attify/firmware-analysis-toolkit/files/580694/FAT_logs_091116.zip)

pabb85, 2016-11-09:
root@kali:~# iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

pabb85, 2016-11-09:
[Diagnosis-FAT-001.txt](https://github.com/attify/firmware-analysis-toolkit/files/580732/Diagnosis-FAT-001.txt)

Then also no response :(

Although the emulated system IP address is resolved to the following mac addy which may be for the emulated system; 00:DE:FA:19:C0:02

arunmagesh, 2016-11-10:
Hey Paul,
Just figure out the issue. 
If you are running it in VM change the network interface to NAT from Bridged. 
It just worked for me perfectly in Kali.

pabb85, 2016-11-10:
Hi Arun,
I just confirmed - was already in Bridged mode, still seeing no ports open :-(

P.

arunmagesh, 2016-11-10:
Change from Bridged mode(This gives you br0) to NAT(this gives you btrunk).

pabb85, 2016-11-10:
Still no :(  even after a reset.sh run...
[image]

I'm getting br0 still (screenshot atrtached)

pabb85, 2016-11-10:
I suspect it is not okay that the "Please check the makeImage function" message is getting echoed...

the function looks like;

```

Def tar2db_and_makeImage(image_id):
	print "Putting information to database"
	try:
		tar2db_command = firmadyne_path + "/scripts/tar2db.py -i " + str(image_id) + " -f " + firmadyne_path + "/images/" + str(image_id) + ".tar.gz"
		output_tar = subprocess.check_output(tar2db_command, shell=True)
		print "Tar2DB" + str(output_tar)
	except:
		print "Already done earlier"
	try:
		print "Creating Image"
		makeImage_command = "sudo " + firmadyne_path + "/scripts/makeImage.sh " + str(image_id)
		print "Executing command \n"
		print str(makeImage_command)
		output_makeImage = subprocess.check_output(makeImage_command, shell=True)
		print "Make Image output " + str(output_makeImage)
	except:
		print "Please check the makeImage function"

```

So, it's taking the exception because some error occured during that block...

P.

pabb85, 2016-11-10:
Aha.....   Right, I may be getting somewhere.

There seem to be some unnecessary commands in the makeImage.sh script where losetup is supposed to remove a loopback device that doesn't exist by that stage, causing the script to 'exit 1' and the fat.py script to print the error message.  

But that doesn't seem to be the problem - and I noticed this on the prior video '003-Firmware analysis - Identifying hardcoded secrets.mp4' where I notice a difference in the results of running binwalk.

So, on the video binwalk unpacks into a nicely structured root but when I do the same, files seem to be misplaced, as if binwalk is no aware of the proper mountpoints for some of the images within the firmware file, for example;

```

root@kali:~/tools/firmadyne# binwalk -e Dlink_firmware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
48            0x30            Unix path: /dev/mtdblock/2
96            0x60            uImage header, header size: 64 bytes, header CRC: 0x7FE9E826, created: 2010-11-23 11:58:41, image size: 878029 bytes, Data Address: 0x80000000, Entry Point: 0x802B5000, data CRC: 0x7C3CAE85, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
160           0xA0            LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2956312 bytes
917600        0xE0060         PackImg section delimiter tag, little endian size: 7348736 bytes; big endian size: 2256896 bytes
917632        0xE0080         Squashfs filesystem, little endian, non-standard signature, version 3.0, size: 2256151 bytes, 1119 inodes, blocksize: 65536 bytes, created: 2010-11-23 11:58:47

root@kali:~/tools/firmadyne# ls -la ./_Dlink_firmware.bin.extracted/
total 18924
drwxr-xr-x 49 root root   36864 Nov 10 15:00 .
drwxr-xr-x 12 root root    4096 Nov 10 15:00 ..
-rw-r--r--  1 root root      17 Nov 23  2010 [
-rw-r--r--  1 root root       0 Nov 23  2010 0
-rw-r--r--  1 root root       0 Nov 23  2010 0ro
-rw-r--r--  1 root root       0 Nov 23  2010 1
-rw-r--r--  1 root root       0 Nov 23  2010 1ro
-rw-r--r--  1 root root       0 Nov 23  2010 2
-rw-r--r--  1 root root       0 Nov 23  2010 2ro
-rw-r--r--  1 root root       0 Nov 23  2010 3
-rw-r--r--  1 root root       0 Nov 23  2010 3ro
-rw-r--r--  1 root root       0 Nov 23  2010 4
-rw-r--r--  1 root root       0 Nov 23  2010 4ro
-rw-r--r--  1 root root       0 Nov 23  2010 5
-rw-r--r--  1 root root       0 Nov 23  2010 5ro
-rw-r--r--  1 root root       0 Nov 23  2010 6
-rw-r--r--  1 root root       0 Nov 23  2010 6ro
-rw-r--r--  1 root root       0 Nov 23  2010 7
-rw-r--r--  1 root root       0 Nov 23  2010 7ro
-rw-r--r--  1 root root 2956312 Nov 10 15:00 A0
-rw-r--r--  1 root root 3174376 Nov 10 15:00 A0.7z
-rw-r--r--  1 root root     395 Nov 23  2010 __ACTION_200.php
-rw-r--r--  1 root root     565 Nov 23  2010 __ACTION_500.php
-rw-r--r--  1 root root    4023 Nov 23  2010 __ACTION.DO.AddPortMapping.php
-rw-r--r--  1 root root    2403 Nov 23  2010 __ACTION.DO.DeletePortMapping.php
-rw-r--r--  1 root root     322 Nov 23  2010 __ACTION.DO.ForceTermination.php
-rw-r--r--  1 root root      93 Nov 23  2010 __ACTION.DO.GetConnectionTypeInfo.php
-rw-r--r--  1 root root      92 Nov 23  2010 __ACTION.DO.GetExternalIPAddress.php
-rw-r--r--  1 root root     425 Nov 23  2010 __ACTION.DO.GetGenericPortMappingEntry.php
-rw-r--r--  1 root root      88 Nov 23  2010 __ACTION.DO.GetNATRSIPStatus.php
-rw-r--r--  1 root root     678 Nov 23  2010 __ACTION.DO.GetSpecificPortMappingEntry.php
-rw-r--r--  1 root root      85 Nov 23  2010 __ACTION.DO.GetStatusInfo.php
-rw-r--r--  1 root root     266 Nov 23  2010 __ACTION.DO.RequestConnection.php
-rw-r--r--  1 root root      22 Nov 23  2010 __ACTION.DO.SetConnectionType.php
-rw-r--r--  1 root root     324 Nov 23  2010 __ACTION.GetCommonLinkProperties.php
-rw-r--r--  1 root root     194 Nov 23  2010 __ACTION.GetConnectionTypeInfo.php
-rw-r--r--  1 root root     212 Nov 23  2010 __ACTION.GetDefaultConnectionService.php

```

Although the normal unix dirs are there;

```

oot@kali:~/tools/firmadyne# ls -la ./_Dlink_firmware.bin.extracted/etc
total 44
drwx------  2 root root  4096 Nov 23  2010 .
drwxr-xr-x 49 root root 36864 Nov 10 15:00 ..
[... 20 lines cut ...]
/usr/lib/python3/dist-packages/binwalk/magic/binwalk: ASCII text
/usr/share/doc/python3-binwalk: directory
/usr/share/doc/binwalk: directory
/usr/bin/binwalk: Python script, ASCII text executable
/root/.config/binwalk: directory
/root/tools/firmware-mod-kit/src/binwalk: directory
/root/tools/binwalk: directory
/root/tools/binwalk/src/scripts/binwalk: Python script, ASCII text executable
/root/tools/binwalk/src/binwalk: directory
/root/tools/binwalk/src/binwalk/magic/binwalk: ASCII text
/root/tools/binwalk/build/scripts-2.7/binwalk: Python script, ASCII text executable
/root/tools/binwalk/build/lib.linux-x86_64-2.7/binwalk: directory
/root/tools/binwalk/build/lib.linux-x86_64-2.7/binwalk/magic/binwalk: ASCII text
root@kali:~/tools/binwalk# /usr/bin/binwalk

Binwalk v2.1.1
Craig Heffner, http://www.binwalk.org
...

```
 
This is v2.1.1. and does the same with the unpacking/extraction...   Which version should we be using please?

Best regards,
Paul

acrixx, 2016-11-10:
Same issue for me too

kali Linux x64 vm

arunmagesh, 2016-11-10:
I don't think "Please check the makeImage function" is an issue. 
Please do check the image.
 [image]
[image]

Note: It may take some time for the network to load the firmware. wait for 2 minutes and open the address in a browser.

pabb85, 2016-11-10:
Hi Arun,

As my previous post - I've identified that the issue appears to be related to how binwalk extracts..

Can you confirm which version works please?

Mine doesn't extract the files in an ordered root like on the videos..

P.

arunmagesh, 2016-11-10:
Mine is V2.1.2b. 

My log:
pastebin.com/Au8f8Ze7

blackdragonth, 2016-11-14:
Hi Arun,

I have the same issue here. But, regardless of interface type of the VM, for me, it occurs with only dlink_firmware.bin
Netgear works fine.

Could you please confirm that your solution works with dlink?

govind-diat, 2016-11-16:
Hi Adi,
   

Dlink and Netgear firmware provided by you is working  fine for me. Right now I am trying to emulate Netgear JNR1010-v1.0.0.10 firmware. While emulation i'm getting the following error and FAT exits

Running the firmware finally : 
**sudo: /home/oit/tools/firmadyne/scratch/1/run.sh: command not found** 

Some more output available in terminal 

Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 3338
Querying database for architecture... mipseb
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 
sudo: /home/oit/tools/firmadyne/scratch/1/run.sh: command not found
Traceback (most recent call last):
  File "./fat.py", line 112, in <module>
    main()
  File "./fat.py", line 108, in main
    final_run(image_id)
  File "./fat.py", line 88, in final_run
    print subprocess.check_output(final_run_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /home/oit/tools/firmadyne/scratch/1/run.sh' returned non-zero exit status 1

In my case no IP address is allotted : Inferring network Interfaces [] is blank

There is FAQ in firmadyne page of github regarding run.sh not getting generated. 

Please provide a solution for same

pabb85, 2016-11-16:
Any update on this please guys?  I beleive the problem is in how different versions of binwalk unpack the various fs files from the firmware images, where some files seem misplaced - as presented above.

As a few people are having an issue with this now and it is putting a blocker on progressing with the course (I have allocated myself time for the course) I have raised with securitytube for further assistance also.

Best regards,
Paul

pabb85, 2016-11-16:
Dear All,

I have took the liberty of diagnosing and fixing the issue, the following are my notes;

*** STARTING FRESH FROM: kali-linux-2016.2-amd64.iso ***

1) Install Kali onto VirtualBox (text based install)

2) Normal post-install actvities (apt update, install guest-additions, etc)

3) Create /root/tools/ and cd into it

4) Proceed to follow steps from: https://github.com/attify/firmware-analysis-toolkit

-> Binwalk already installed (v2.1.1) so skipping binwalk install for now

5) Setting up firmadyne;
a) sudo apt-get install busybox-static fakeroot git kpartx netcat-openbsd nmap python-psycopg2 python3-psycopg2 snmp uml-utilities util-linux vlan qemu-system-arm qemu-system-mips qemu-system-x86 qemu-utils
b) git clone --recursive https://github.com/firmadyne/firmadyne.git
c) cd ./firmadyne; ./download.sh
d) Edit firmadyne.config and make the FIRMWARE_DIR point to /root/tools/firmadyne/
e) cd ..

6) Setting up Firmware-mod-Kit;
a) sudo apt-get install git build-essential zlib1g-dev liblzma-dev python-magic
b) git clone https://github.com/brianpow/firmware-mod-kit
c) Modify the file shared-ng.inc to change the value of variable BINWALK to /usr/bin/binwalk
d) cd ..

-> MITMProxy already installed, skipping this

7) Setting up Firmwalker
a) git clone https://github.com/craigz28/firmwalker.git 

8) Setting up Firmware Analysis Toolkit;
a) git clone https://github.com/attify/firmware-analysis-toolkit.git
b) mv ./firmware-analysis-toolkit/* ./firmadyne/; rm -rf ./firmware-analysis-toolkit/
c) cd firmadyne; chmod +x *.py *.sh

9) Setting up Postgres (already installed but needs configuring);
a) update-rc.d postgresql enable
b) service postgresql start
c) sudo -u postgres createuser firmadyne
d) sudo -u postgres psql 
   ...
   postgres=# \password firmadyne
   Enter new password: 
   Enter it again: 
   postgres-# \q
e) ./reset.sh 
f) cd ..

<- Trying to run fat.py here fails, complains about binwalk module being missing and installing binwalk python module via pip doesn't fix the issue so going to steps from the setup instructions, after removing the default kali version of binwalk

10) Install Binwalk;
a) apt-get remove binwalk
b) git clone https://github.com/devttys0/binwalk.git
c) cd binwalk

<- deps.sh would fail here because some packages aren't included in the kali repos, so remove cramfsprogs, cramfsswap and openjdk-7-jdk on line 16 of deps.sh before running it and then manually install the missing cramfs packages (openjdk-8-jdk is already installed)

d) ./deps.sh
e) wget http://mirrors.kernel.org/ubuntu/pool/universe/c/cramfsswap/cramfsswap_1.4.1ubuntu1_amd64.deb
f) wget http://security.ubuntu.com/ubuntu/pool/universe/c/cramfs/cramfsprogs_1.1-6ubuntu1_amd64.deb
g) dpkg -i *.deb
h) python ./setup.py install
i) apt-get install python-lzma
j) sudo -H pip install git+https://github.com/ahupp/python-magic
k) cd ..

Now fat.py works for me for the wnap320.zip firmware!
[image]

pabb85, 2016-11-16:
Arrrgghhh.....   
Well, I thought it was fine but.....  
When I now try and run fat.py with the Dlink_firmware.bin it fails.  

With wnap320.zip works still but when I try to use the Dlink_firmware.bin with every setting the *exact* same (NAT, etc) - it gives the following;

[image]

Back to the drawing board...

blackdragonth, 2016-11-16:
Hi pabb,

Same as me. And if you take a look at qeme.finalize... log it keeps getting bigger and bigger.
Looks like it is trying to do something again and again, which I am currently figure it out.

And our instructure seems to be unaware of that when I passed him my log.

pabb85, 2016-11-16:
Strange stuff hey...  I'm wondering if the right Dlink firmware file got included in the course pack?

I just tried a few things including changing the 60s wait to 180s but no success yet.  I suspect the issue is related to how the Dlink_firmware.bin file is unpacked by binwalk, because;

```
root@iot-kali:~/lab/Dlink_firmware# binwalk 

Binwalk v2.1.2b
Craig Heffner, http://www.binwalk.org
...
root@iot-kali:~/lab/Dlink_firmware# binwalk -e Dlink_firmware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
48            0x30            Unix path: /dev/mtdblock/2
96            0x60            uImage header, header size: 64 bytes, header CRC: 0x7FE9E826, created: 2010-11-23 11:58:41, image size: 878029 bytes, Data Address: 0x80000000, Entry Point: 0x802B5000, data CRC: 0x7C3CAE85, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
160           0xA0            LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2956312 bytes
917600        0xE0060         PackImg section delimiter tag, little endian size: 7348736 bytes; big endian size: 2256896 bytes
917632        0xE0080         Squashfs filesystem, little endian, non-standard signature, version 3.0, size: 2256151 bytes, 1119 inodes, blocksize: 65536 bytes, created: 2010-11-23 11:58:47

root@iot-kali:~/lab/Dlink_firmware# cd _Dlink_firmware.bin.extracted/
root@iot-kali:~/lab/Dlink_firmware/_Dlink_firmware.bin.extracted# ls -la
total 18924
drwxr-xr-x 49 root root   36864 Nov 16 19:16 .
drwxr-xr-x  3 root root    4096 Nov 16 19:16 ..
-rw-r--r--  1 root root      17 Nov 23  2010 [
-rw-r--r--  1 root root       0 Nov 23  2010 0
-rw-r--r--  1 root root       0 Nov 23  2010 0ro
-rw-r--r--  1 root root       0 Nov 23  2010 1
-rw-r--r--  1 root root       0 Nov 23  2010 1ro
-rw-r--r--  1 root root       0 Nov 23  2010 2
-rw-r--r--  1 root root       0 Nov 23  2010 2ro
-rw-r--r--  1 root root       0 Nov 23  2010 3
-rw-r--r--  1 root root       0 Nov 23  2010 3ro
-rw-r--r--  1 root root       0 Nov 23  2010 4
-rw-r--r--  1 root root       0 Nov 23  2010 4ro
-rw-r--r--  1 root root       0 Nov 23  2010 5
-rw-r--r--  1 root root       0 Nov 23  2010 5ro
-rw-r--r--  1 root root       0 Nov 23  2010 6
-rw-r--r--  1 root root       0 Nov 23  2010 6ro
-rw-r--r--  1 root root       0 Nov 23  2010 7
-rw-r--r--  1 root root       0 Nov 23  2010 7ro
-rw-r--r--  1 root root 2956312 Nov 16 19:16 A0
-rw-r--r--  1 root root 3174376 Nov 16 19:16 A0.7z
-rw-r--r--  1 root root     395 Nov 23  2010 __ACTION_200.php
-rw-r--r--  1 root root     565 Nov 23  2010 __ACTION_500.php
-rw-r--r--  1 root root    4023 Nov 23  2010 __ACTION.DO.AddPortMapping.php
-rw-r--r--  1 root root    2403 Nov 23  2010 __ACTION.DO.DeletePortMapping.php
-rw-r--r--  1 root root     322 Nov 23  2010 __ACTION.DO.ForceTermination.php
-rw-r--r--  1 root root      93 Nov 23  2010 __ACTION.DO.GetConnectionTypeInfo.php
-rw-r--r--  1 root root      92 Nov 23  2010 __ACTION.DO.GetExternalIPAddress.php
-rw-r--r--  1 root root     425 Nov 23  2010 __ACTION.DO.GetGenericPortMappingEntry.php
-rw-r--r--  1 root root      88 Nov 23  2010 __ACTION.DO.GetNATRSIPStatus.php
-rw-r--r--  1 root root     678 Nov 23  2010 __ACTION.DO.GetSpecificPortMappingEntry.php
-rw-r--r--  1 root root      85 Nov 23  2010 __ACTION.DO.GetStatusInfo.php
-rw-r--r--  1 root root     266 Nov 23  2010 __ACTION.DO.RequestConnection.php
-rw-r--r--  1 root root      22 Nov 23  2010 __ACTION.DO.SetConnectionType.php
-rw-r--r--  1 root root     324 Nov 23  2010 __ACTION.GetCommonLinkProperties.php
-rw-r--r--  1 root root     194 Nov 23  2010 __ACTION.GetConnectionTypeInfo.php
-rw-r--r--  1 root root     212 Nov 23  2010 __ACTION.GetDefaultConnectionService.php
-rw-r--r--  1 root root     136 Nov 23  2010 __ACTION.GetEthernetLinkStatus.php
-rw-r--r--  1 root root     184 Nov 23  2010 __ACTION.GetExternalIPAddress.php
-rw-r--r--  1 root root     665 Nov 23  2010 __ACTION.GetGenericPortMappingEntry.php
-rw-r--r--  1 root root      86 Nov 23  2010 __ACTION.GetNATRSIPStatus.php
-rw-r--r--  1 root root     378 Nov 23  2010 __ACTION.GetSpecificPortMappingEntry.php
-rw-r--r--  1 root root     433 Nov 23  2010 __ACTION.GetStatusInfo.php
-rw-r--r--  1 root root     146 Nov 23  2010 __ACTION.GetTotalBytesReceived.php
-rw-r--r--  1 root root     138 Nov 23  2010 __ACTION.GetTotalBytesSent.php
-rw-r--r--  1 root root     152 Nov 23  2010 __ACTION.GetTotalPacketsReceived.php
-rw-r--r--  1 root root     144 Nov 23  2010 __ACTION.GetTotalPacketsSent.php
[... 26 lines cut ...]
-rw-r--r--  1 root root    1575 Nov 23  2010 adv_wlan.php
-rw-r--r--  1 root root     328 Nov 23  2010 __ajax_tools_ddns_info.php
-rw-r--r--  1 root root     307 Nov 23  2010 __ajax_tools_ddns_setnodes.php
-rw-r--r--  1 root root     480 Nov 23  2010 __ajax_wan_conninfo.php
drwx------  2 root root    4096 Nov 23  2010 alt
-rw-r--r--  1 root root    1134 Nov 23  2010 application_rule.xml
-rw-r--r--  1 root root       7 Nov 23  2010 arpping
-rw-r--r--  1 root root       7 Nov 23  2010 asession
-rw-r--r--  1 root root    9900 Nov 23  2010 ated
-rw-r--r--  1 root root   46856 Nov 23  2010 atp
-rw-r--r--  1 root root     107 Nov 23  2010 attack.tmp
drwx------  2 root root    4096 Nov 23  2010 auth
-rw-r--r--  1 root root     590 Nov 23  2010 __auth_check.php
-rw-r--r--  1 root root    1663 Nov 23  2010 __authenticate_p.php
[... cut ...]
So what I am seeing is that some of the parts of the FS are getting unpacked directly into the root when they should surely be within their own place on the filesystem - this would explain to me why the emulated image appears to come up (nmap finds a mac address for it) but the services aren't started properly (the FS is messed up and config files etc are misplaced).

I really hope the Attify/SecurityTube guys can help a bit quicker on this because I had allocated myself time to work through this course and the time I'm losing on trying to get the setup working is way too much...   *sigh*

blackdragonth, 2016-11-16:
@pabb85
Agreed.

pabb85, 2016-11-16:
@blackdragonth Are your videos numbered correctly?  I found that video 5 seemed to be referencing things talked about in video 7 as if they were already covered.  

I've probably bought every ST course they've made and never had issues like this before :-(

adi0x90, 2016-11-16:
Thanks @pabb85 for the description. 

@blackdragonth @acrixx  : 
To provide a quick solution, here's the most recent build of **AttifyOS** which you could use, which will work with all the firmware emulation part of the course and the other exercises (_everything except only the Zigbee exploitation_) : 

Download - [Dropbox Link - http://tinyurl.com/attifyos](http://tinyurl.com/attifyos) 

**Note:** _The file has been compressed using 7-zip and might need 7-zip or WinRar on Windows / other decompression software on Mac for it to decompress. Let me know if you still face any issues on the VM._ 
 
----

In the meanwhile, I'm looking into the issue and this most probably looks like a sasquatch or unsquashfs as devttys0 (Binwalk author) pointed out. However, I'm trying to see if I could get a fix ready for this and share with all of you.

pabb85, 2016-11-16:
@adi0x90 Thank eatyou!

The internet connection where I am isn't so great but I have started to download the AttifyOS files - maybe they'll be finished by tomorrow :-)

If you need to emulate where we are at - my steps above will bring you to the exact state of the problem, starting from the latest Kali 2016.2 x86_64 ISO because they're exactly as I did earlier on a fresh install.

Best regards,
P.

pabb85, 2016-11-16:
@adi0x90 

This issue may be connected;

```
root@iot-kali:~/tools/firmware-mod-kit# ./extract-firmware.sh ./Dlink_firmware.bin 
Firmware Mod Kit (extract) 0.99, (c)2011-2013 Craig Heffner, Jeremy Collake

Preparing tools ...
bff_huffman_decompress.c: In function ‘unpack_parse_header’:
bff_huffman_decompress.c:167:14: warning: implicit declaration of function ‘read’ [-Wimplicit-function-declaration]
  bytesread = read(in, hdr + prelen, PACK_HEADER_LENGTH - prelen);
              ^~~~
bff_huffman_decompress.c: In function ‘unpack’:
bff_huffman_decompress.c:318:22: warning: implicit declaration of function ‘dup’ [-Wimplicit-function-declaration]
  unpack_parse_header(dup(in), dup(out), pre, prelen, bytes_in, &unpackd);
                      ^~~
mksquashfs.o: In function `linux_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1384: undefined reference to `add_dir_entry'
mksquashfs.o: In function `encomp_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1399: undefined reference to `add_dir_entry'
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1416: undefined reference to `add_dir_entry'
mksquashfs.o: In function `single_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1431: undefined reference to `add_dir_entry'
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1451: undefined reference to `add_dir_entry'
collect2: error: ld returned 1 exit status
Makefile:18: recipe for target 'mksquashfs' failed
make[2]: *** [mksquashfs] Error 1
Makefile:2: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 2
Tools build failed! Check pre-requisites. Quitting...
```

adi0x90, 2016-11-17:
@pabb85 : This does not look like a issue connected to FAT, but rather with Firmware-mod-kit. Firmware-mod-kit github project has not been active for quite a while. 

However, since this is something that we do in the course, can you open up a separate issue for this so that other students can also benefit from the same. 

Also, in the new issue, please check and mention the version of mksquashfs or see if squashfs-tools is correctly installed.

govind-diat, 2016-11-17:
Hi guys,

Issue 1
I'm getting following error message while trying to emulate netgear jnr1010v2 firmwares.

Running the firmware finally :
sudo: /home/oit/tools/firmadyne/scratch/1/run.sh: command not found

Issue 2
While emulating firmware for dlink DIR-601, linksys1200E, DSL2730U no errors are displayed.
However when open browser, webpages are not loaded for firmwares. On running nmap it shows all the ports closed.

Anyone facing such issues, pls help

adi0x90, 2016-11-17:
@govind-diat : 

Can you share the firmware files so that I can have a look. 

Also, as mentioned in the firmadyne [FAQ](https://github.com/firmadyne/firmadyne#faq) this could be because of the following reasons : 

[image]

In case if it is any of those above, I would recommend you to post an issue in [Firmadyne github repo](https://github.com/firmadyne/firmadyne/issues) , so that the authors would be able to fix this if this is a bug in Firmadyne or in case you need some modification to make it work.

mgaurav5, 2016-11-19:
So with the attifyOS, WNAP320 worked without issues, but still having issues with Dlink_firmware.bin. Like someone mentioned earlier, could it be an issue with the firmware copy?

Password for user firmadyne: 
qemu: terminating on signal 2 from pid 10418
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 
sudo: /home/oit/tools/fat//scratch/1/run.sh: command not found
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 109, in main
    final_run(image_id)
  File "./fat.py", line 89, in final_run
    print subprocess.check_output(final_run_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /home/oit/tools/fat//scratch/1/run.sh' returned non-zero exit status 1

ankushgoel27, 2016-11-19:
netgear firmware works fine however the dlink firmware gives an ip of 192.168.0.1 and the web browser cant reach to this ip.

adi0x90, 2016-11-21:
@mgaurav5 @ankushgoel27 Yes, I think the issue might be with the Dlink firmware then. Let me again check the firmware and I'll update.

pabb85, 2016-11-21:
Hi chaps - I can report success using the AttifyOS with the Dlink_firmware 
[image]

adi0x90, 2016-11-22:
@pabb85 : Glad to hear. Thanks for confirming. 

Closing the issue for now. Will update the README once there is a solution available for running everything on Kali Linux.

closed 2016-11-22 by adi0x90

zvm91, 2019-08-04:
Hi,
@adi0x90 I’m working on firmware emulation with firmadyne and I had the same error as irini90. I use your automate script that you posted before and I got this error:

 ./scripts/getArch.sh ./images/1.tar.gz
tar: ./images/1.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now

can you help me please?