attify/firmware-analysis-toolkit #2: Firmware-mod-kit issues under Kali

pabb85, 2016-11-17:
@adi0x90 

Issue when running firmware-mod-kit (it goes to compile the tools on first run);

```
root@iot-kali:~/tools/firmware-mod-kit# ./extract-firmware.sh ./Dlink_firmware.bin 
Firmware Mod Kit (extract) 0.99, (c)2011-2013 Craig Heffner, Jeremy Collake

Preparing tools ...
bff_huffman_decompress.c: In function ‘unpack_parse_header’:
bff_huffman_decompress.c:167:14: warning: implicit declaration of function ‘read’ [-Wimplicit-function-declaration]
  bytesread = read(in, hdr + prelen, PACK_HEADER_LENGTH - prelen);
              ^~~~
bff_huffman_decompress.c: In function ‘unpack’:
bff_huffman_decompress.c:318:22: warning: implicit declaration of function ‘dup’ [-Wimplicit-function-declaration]
  unpack_parse_header(dup(in), dup(out), pre, prelen, bytes_in, &unpackd);
                      ^~~
mksquashfs.o: In function `linux_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1384: undefined reference to `add_dir_entry'
mksquashfs.o: In function `encomp_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1399: undefined reference to `add_dir_entry'
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1416: undefined reference to `add_dir_entry'
mksquashfs.o: In function `single_opendir':
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1431: undefined reference to `add_dir_entry'
/root/tools/firmware-mod-kit/src/others/squashfs-2.2-r2-7z/mksquashfs.c:1451: undefined reference to `add_dir_entry'
collect2: error: ld returned 1 exit status
Makefile:18: recipe for target 'mksquashfs' failed
make[2]: *** [mksquashfs] Error 1
Makefile:2: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 2
Tools build failed! Check pre-requisites. Quitting...
```

pabb85, 2016-11-17:
I have this working on my system now by removing the offending item from the ~/tools/firmware-mod-kit/src/others/Makefile - as mksquashfs was already installed on my system, there's no point trying (failing) to build it again.

Also see: https://bitsum.com/forum/index.php?topic=2018.0 - You may need a few extra dependencies.

adi0x90, 2016-11-17:
Perfect. Thanks for resolving this @pabb85 .

closed 2016-11-17 by adi0x90

BecauseTheWorldIsRound, 2025-01-27:
On another project, I had the same problem.
I noticed that `add_dir_entry` is declared `inline`.
Depending on the compiler version, an `inline` function may not be exported. 
Removing `inline` fixed the problem.