attify/firmware-analysis-toolkit #4: FAT issue

ankushgoel27, 2016-11-18:
root@kali:~/Downloads/IOT-tools/firmware-analysis-toolkit# python fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : /root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin
Enter the brand of the firmware : DLink
/root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin
Now going to extract the firmware. Hold on..
/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b DLink -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin" images 
Traceback (most recent call last):
  File "/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py", line 730, in <module>
    main()
  File "/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py", line 727, in main
    extract.extract()
  File "/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py", line 189, in extract
    self._extract_item(item)
  File "/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py", line 197, in _extract_item
    ExtractionItem(self, path, 0).extract()
  File "/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py", line 227, in __init__
    host=self.extractor.database)
  File "/usr/lib/python2.7/dist-packages/psycopg2/__init__.py", line 164, in connect
    conn = _connect(dsn, connection_factory=connection_factory, async=async)
psycopg2.OperationalError: could not connect to server: Connection refused
	Is the server running on host "127.0.0.1" and accepting
	TCP/IP connections on port 5432?

Exception AttributeError: "'ExtractionItem' object has no attribute 'database'" in <bound method ExtractionItem.__del__ of <__main__.ExtractionItem object at 0x7f48f838e710>> ignored
Traceback (most recent call last):
  File "fat.py", line 113, in <module>
    main()
  File "fat.py", line 100, in main
    extractor(firm_name,firm_brand)
  File "fat.py", line 45, in extractor
    output = subprocess.check_output(extractor_command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b DLink -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin" images ' returned non-zero exit status 1

adi0x90, 2016-11-18:
Try `service postgressql start`

ankushgoel27, 2016-11-19:
i am getting another error now

nter the name or absolute path of the firmware you want to analyse : /root/Downloads/IOT-tools/exercisefiles/firmware
Enter the brand of the firmware : kkeps.bin
/root/Downloads/IOT-tools/exercisefiles/firmware
Now going to extract the firmware. Hold on..
/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b kkeps.bin -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware" images 
test
The database ID is 3
test
The database ID is 4
test
The database ID is 5
test
The database ID is 6
test
The database ID is 1
test
The database ID is 7
test
The database ID is 2
test
The database ID is 8
test
The database ID is 9
test
The database ID is 0
test
The database ID is 1
test
The database ID is 2
Getting image type
Traceback (most recent call last):
  File "fat.py", line 113, in <module>
    main()
  File "fat.py", line 105, in main
    get_image_type(image_id)
  File "fat.py", line 58, in get_image_type
    output = subprocess.check_output(get_image_type_command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/root/Downloads/IOT-tools/firmadyne/scripts/getArch.sh ./images/2.tar.gz' returned non-zero exit status 1

ankushgoel27, 2016-11-19:
i copied the firmadyne.config file into the firmware-analysis-toolkit folder and this error seems to have gone away. however here are few more errors. it seems like its getting the wrong image id. 
****
root@kali:~/Downloads/IOT-tools/firmware-analysis-toolkit# python fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : /root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin
Enter the brand of the firmware : Dlink
/root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin
Now going to extract the firmware. Hold on..
/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware/Dlink_firmware.bin" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Traceback (most recent call last):
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 100, in <module>
    main()
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 68, in process
    (files, links) = getFileHashes(infile)
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 11, in getFileHashes
    t = tarfile.open(infile)
  File "/usr/lib/python2.7/tarfile.py", line 1673, in open
    return func(name, "r", fileobj, **kwargs)
  File "/usr/lib/python2.7/tarfile.py", line 1738, in gzopen
    fileobj = gzip.GzipFile(name, mode, compresslevel, fileobj)
  File "/usr/lib/python2.7/gzip.py", line 94, in __init__
    fileobj = self.myfileobj = __builtin__.open(filename, mode or 'rb')
IOError: [Errno 2] No such file or directory: '/root/Downloads/IOT-tools/firmadyne/images/1.tar.gz'
Already done earlier
Creating Image
Executing command 

sudo /root/Downloads/IOT-tools/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu-system-mipsel: -drive if=ide,format=raw,file=/root/Downloads/IOT-tools/firmadyne/scratch//1//image.raw: Could not open '/root/Downloads/IOT-tools/firmadyne/scratch//1//image.raw': No such file or directory
Traceback (most recent call last):
  File "fat.py", line 113, in <module>
    main()
  File "fat.py", line 108, in main
    network_setup(image_id)
  File "fat.py", line 83, in network_setup
    output = subprocess.check_output(network_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /root/Downloads/IOT-tools/firmadyne/scripts/inferNetwork.sh 1' returned non-zero exit status 1

ankushgoel27, 2016-11-19:
please close this issue. Didnt realize that the fat.py needs to be inside the firmadyne directory.

ankushgoel27, 2016-11-19:
the script is working fine now but i am not able to reach the ip address generated for the device.
root@kali:~/Downloads/IOT-tools/firmadyne# python fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : /root/Downloads/IOT-tools/exercisefiles/firmware/kkeps.bin
Enter the brand of the firmware : kkeps
/root/Downloads/IOT-tools/exercisefiles/firmware/kkeps.bin
Now going to extract the firmware. Hold on..
/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b kkeps -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware/kkeps.bin" images 
test
The database ID is 4
Getting image type
Password for user firmadyne: 
Found image type of  mipseb
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /root/Downloads/IOT-tools/firmadyne/scripts/makeImage.sh 4
Password for user firmadyne: 
mke2fs 1.43.3 (04-Sep-2016)
mknod: /dev/console: File exists
Please check the makeImage function
Everything is done for the image id 4
Setting up the network connection
Password for user firmadyne: 
qemu-system-mips: terminating on signal 2 from pid 1957
Querying database for architecture... mipseb
Running firmware 4: terminating after 60 secs...
Bad SWSTYLE=0x04
Inferring network...
Interfaces: [('eth0', '192.168.1.1'), ('br-lan', '192.168.10.253')]
Done!

Running the firmware finally :

Not able to ping or http to either of the addresses above. Please help

ankushgoel27, 2016-11-19:
and now i am getting this error. LOL
root@kali:~/Downloads/IOT-tools/firmadyne# python fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : /root/Downloads/IOT-tools/exercisefiles/firmware/WNDR4700-V1.0.0.52.zip
Enter the brand of the firmware : wndr
/root/Downloads/IOT-tools/exercisefiles/firmware/WNDR4700-V1.0.0.52.zip
Now going to extract the firmware. Hold on..
/root/Downloads/IOT-tools/firmadyne/sources/extractor/extractor.py -b wndr -sql 127.0.0.1 -np -nk "/root/Downloads/IOT-tools/exercisefiles/firmware/WNDR4700-V1.0.0.52.zip" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Traceback (most recent call last):
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 100, in <module>
    main()
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 77, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "/root/Downloads/IOT-tools/firmadyne/scripts/tar2db.py", line 61, in insertObjectToImage
    for x in links])
psycopg2.IntegrityError: duplicate key value violates unique constraint "object_to_image_oid_iid_filename_key"
DETAIL:  Key (oid, iid, filename)=(1, 1, /bin/echo) already exists.

Already done earlier
Creating Image
Executing command 

sudo /root/Downloads/IOT-tools/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
mke2fs 1.43.3 (04-Sep-2016)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu-system-mipsel: terminating on signal 2 from pid 1846
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Done!

Running the firmware finally :

adi0x90, 2016-11-22:
The above message indicates that there is a conflicting database entry for the same ID. Run `reset.sh` to clear the database and run the `fat.py` again.

ankushgoel27, 2016-11-22:
i did. but i still cant access that IP in the browser.

ankushgoel27, 2016-11-22:
i think you can close this issue. This is the same one that you closed earlier for kali linux not emulating the Dlink firmware.

closed 2016-11-28 by adi0x90