attify/firmware-analysis-toolkit #6: Unable to emulate repacked firmware(using firmware mod kit) on fat

nitmalviya03, 2017-04-11:
Hello. When I try to emulate the repacked firmware using fmk, I am unable to emulate the firmware. It shows following errors at the end

root@ubuntu:/home/oit/tools/fat# ./fat.py 

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : new-firmware.bin
Enter the brand of the firmware : netgear
new-firmware.bin
Now going to extract the firmware. Hold on..
/home/oit/tools/fat//sources/extractor/extractor.py -b netgear -sql 127.0.0.1 -np -nk "new-firmware.bin" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /home/oit/tools/fat//scripts/makeImage.sh 1
Password for user firmadyne: 
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0xd60a0292.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
Building a new DOS disklabel with disk identifier 0xc370d1a5.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.9 (4-Feb-2014)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 4220
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 
sudo: /home/oit/tools/fat//scratch/1/run.sh: command not found
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 109, in main
    final_run(image_id)
  File "./fat.py", line 89, in final_run
    print subprocess.check_output(final_run_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /home/oit/tools/fat//scratch/1/run.sh' returned non-zero exit status 1

I am running official ubuntu vm provided via dropbox link

nitmalviya03, 2017-04-12:
Hey it worked fine.I was making a small mistake while building the firmware

closed 2017-04-12 by nitmalviya03

Varbaek, 2017-08-12:
@nitmalviya03 Can you tell us what the small mistake was?

nitmalviya03, 2017-08-12:
@Varbaek i wasn't entering / after giving the path to the firmware file