attify/firmware-analysis-toolkit #7: Can you help in this error

HackB4sEcurE, 2017-06-12:
iot@I:~/firmware-analysis-toolkit$ sudo ./fat.py 

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : DIR.bin
Enter the brand of the firmware : Dlink
DIR830LA1_FW100B07.bin
```
Now going to extract the firmware. Hold on..
/home/iot/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "DIR830LA1_FW100B07.bin" images 
Traceback (most recent call last):
  File "/home/iot/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py", line 17, in <module>
    import magic
ImportError: No module named magic
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 100, in main
    extractor(firm_name,firm_brand)
  File "./fat.py", line 45, in extractor
    output = subprocess.check_output(extractor_command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/home/iot/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "DIR830LA1_FW100B07.bin" images ' returned non-zero exit status 1
```

sgript, 2017-06-14:
Having the same issue also.

extremecoders-re, 2018-03-23:
Closing due to lack of information and spam comments.

closed 2018-03-23 by extremecoders-re