attify/firmware-analysis-toolkit #12: Emulation run fail

juushya, 2017-11-16:
Hi,

The emulation fails with the following errors:

```
...
 [+] Cleaning previous images and created files by firmadyne
 [+] All done. Go ahead and run fat.py to continue firmware analysis
     
 Remember the password for the database is firmadyne

/home/oit/tools/firmadyne [git::master *] [oit@ubuntu] [15:19]
> sudo ./fat.py 

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : /home/oit/private/private.bin
Enter the brand of the firmware : private
/home/oit/private/private.bin
Now going to extract the firmware. Hold on..
/home/oit/tools/firmadyne/sources/extractor/extractor.py -b private -sql 127.0.0.1 -np -nk "/home/oit/private/private.bin" images 
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /home/oit/tools/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0xc7513c37.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
Building a new DOS disklabel with disk identifier 0x0d2d9eb9.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.9 (4-Feb-2014)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 8721
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 
sudo: /home/oit/tools/firmadyne/scratch/1/run.sh: command not found
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 109, in main
    final_run(image_id)
  File "./fat.py", line 89, in final_run
    print subprocess.check_output(final_run_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /home/oit/tools/firmadyne/scratch/1/run.sh' returned non-zero exit status 1

/home/oit/tools/firmadyne [git::master *] [oit@ubuntu] [15:21]
> 
```
AttifyOS VM is running in VMWare Fusion, with one NIC eth0 (tried with NAT, Bridged, Host-Only, separately)..

extremecoders-re, 2017-11-20:
The issue you are facing is because firmadyne was not able configure the network properly.

You can try out the steps listed [here](https://github.com/firmadyne/firmadyne#runsh-is-not-generated).  In particular try increasing the timeout in `inferNetwork.sh`

extremecoders-re, 2018-01-27:
Closing. If you need further assistance, feel free to reopen.

closed 2018-01-27 by extremecoders-re

bharat089, 2018-02-06:
I am using firmadyne to run firmware. It is working for given example. When I tried to run different firmware (i.e. Dlink firmware- http://drivers.softpedia.com/dyn-postdownload.php/9619c307d433b497a1104954dad4f182/5a797107/7a86a/0/1) , It worked till last step and then I am getting this [ 2323.940000 ] firmadyne: ioctl: 0x3] the first number keeps on increasing. Could you please help me to understand this?

extremecoders-re, 2018-02-07:
@bharat089 
Not all firmwares can be emulated with firmadyne specifically those which requires access to peripheral devices absent in the emulated environment.

Now talking about this line `[ 2323.940000 ] firmadyne: ioctl: 0x3]` , the first number in square braces is the timestamp. It indicates the time elapsed in seconds since the kernel booted up. When something on the system issues a IOCTL which firmadyne doesn't know how to emulate, you will get that message.

bharat089, 2018-02-07:
Thanks for the clarification..!!!

Another problem, I am facing while emulating asus router firmware, It stucks in "[ 31.837692 ] cfg 80211 calling crda to update world regulatory domain " and then it says at last "cfg80211 exceeded crda call max attempts. Not calling CRDA". What is this message about?

extremecoders-re, 2018-02-07:
@bharat089 The message is likely related to the cfg80211 kernel module which provides WiFi functionality.