attify/firmware-analysis-toolkit #13: pexpect.exceptions.EOF

tohaikmeng, 2018-01-13:
Hi,

anyone experienced the error below?

Traceback (most recent call last):
  File "fat.py", line 122, in <module>
    main()
  File "fat.py", line 109, in main
    image_id = run_extractor(firm_name, firm_brand)
  File "fat.py", line 48, in run_extractor
    child.expect("Database Image ID: ")
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 327, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 355, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 102, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 49, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fcc42f568d0>
command: /home/who/firmadyne/sources/extractor/extractor.py
args: ['/home/who/firmadyne/sources/extractor/extractor.py', '-b', 'aztech', '-sql', '127.0.0.1', '-np', '-nk', 'firmware.w', 'images']
buffer (last 100 chars): ''
before (last 100 chars): 'bound method ExtractionItem.__del__ of <__main__.ExtractionItem object at 0x7f0cdda3a050>> ignored\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 32431
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Database Image ID: ")

any advices is greatly appreciated. :-)

extremecoders-re, 2018-01-13:
FAT uses firmadyne which internally uses binwalk.

The error message you are getting is due to extractor.py component of firmadyne failing. 
Can you run just binwalk to check if this is indeed the case.

```
binwalk -e <firmware file>
```

Alternatively, you can just run extractor.py as below.

```
./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320 Firmware Version 2.0.3.zip" images
```

extremecoders-re, 2018-01-27:
Closing. If your issue persists, feel free to reopen.

closed 2018-01-27 by extremecoders-re

mgp25, 2018-07-18:
@extremecoders-re sorry to revive this issue, but I was facing the same issue as detailed above. I tried your suggestion, both:

`binwalk -e <firmware file>`

or  

`./sources/extractor/extractor.py -b d -sql 127.0.0.1 -np -nk WNAP320 FW.bin images`

Did not throw any error.

Any idea what could be the issue?

Regards

jpc0016, 2019-03-13:
Good afternoon,

I experience the same error.  I am able to run your extractor.py command but get this warning when running binwalk

`WARNING: Extractor.execute failed to run external extractor 'jar xvf '%e'': [Errno 2] No such file or directory, 'jar xvf '%e'' might not be installed correctly`

`Traceback (most recent call last):
  File "fat.py", line 123, in <module>
    main()
  File "fat.py", line 110, in main
    image_id = run_extractor(firm_name, firm_brand)
  File "fat.py", line 49, in run_extractor
    child.expect("Database Image ID: ")
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f367cedbf10>
command: /home/jpc0016/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py
args: ['/home/jpc0016/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py', '-b', 'netgear', '-sql', '127.0.0.1', '-np', '-nk', 'WNAP320_v2.0.3.zip', 'images']
buffer (last 100 chars): ''

after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 19764
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('Database Image ID: ')`