attify/firmware-analysis-toolkit #14: Huawei Firmware Extraction

alivelimeli, 2018-01-24:
Hello @adi0x90 , @extremecoders-re 

I'm trying to identify a Huawei modem firmware binary using binwalk but it doesn't recognize it as jiffs, squash etc

```
$ binwalk -v Huawei.bin 

Scan Time:     2018-01-22 21:34:10
Target File:   /media/data/tmp/Huawei.bin
MD5 Checksum:  e1ddf1d896631b07331b5188d5b31ca2
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
4992299       0x4C2D2B        MySQL ISAM index file Version 1
```

It's definitely not MySQL ISAM index file :)

So, how can I go further? Is there an option to force binwalk to extract it as jiffs file system or squash?

Here is the file:
 
[Huawei.bin.zip](https://github.com/ReFirmLabs/binwalk/files/1653158/Huawei.bin.zip)

binwalk opcodes result:
```
$ binwalk -A Huawei.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
12713598      0xC1FE7E        ARM instructions, function prologue
16463970      0xFB3862        ARM instructions, function prologue

```

Entropy:
```
$ binwalk -E Huawei.bin

DECIMAL       HEXADECIMAL     ENTROPY
--------------------------------------------------------------------------------
0             0x0             Rising entropy edge (0.996797)

```
[image]

So, how can I go further? Is there any other software or suggestion in firmware-analysis-kit?

extremecoders-re, 2018-01-24:
Hi, 
The entropy scans suggest that firmware is encrypted. There is no proper header or trailer for format identification. In such cases binwalk cannot do much. I would suggest to search on Google about the specific model of the device.

extremecoders-re, 2018-01-27:
Closing this. If you need further assistance, feel free to reopen.

closed 2018-01-27 by extremecoders-re