attify/firmware-analysis-toolkit #16: pexpect tripping during extraction

jkbecker, 2018-04-01:
Hi,
I installed the firmware analysis toolkit and it will die as show below -- I'm showing the example device image from your instructions but can also reproduce this using other images.

I'm running this on an up to date Kali Linux.

``` 

                               __           _   
                              / _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : /root/Desktop/WNAP320 Firmware Version 2.0.3.zip
[?] Enter the brand of the firmware : netgear
[+] Now going to extract the firmware. Hold on..
[+] Firmware : /root/Desktop/WNAP320 Firmware Version 2.0.3.zip
[+] Brand : netgear
Traceback (most recent call last):
  File "./fat.py", line 122, in <module>
    main()
  File "./fat.py", line 109, in main
    image_id = run_extractor(firm_name, firm_brand)
  File "./fat.py", line 48, in run_extractor
    child.expect("Database Image ID: ")
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7ff20f3ff410>
command: /root/device-simulator/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py
args: ['/root/device-simulator/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py', '-b', 'netgear', '-sql', '127.0.0.1', '-np', '-nk', '/root/Desktop/WNAP320 Firmware Version 2.0.3.zip', 'images']
buffer (last 100 chars): ''
before (last 100 chars): 'bound method ExtractionItem.__del__ of <__main__.ExtractionItem object at 0x7effcf4f5e90>> ignored\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 4899
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Database Image ID: ")
```

r00tb3, 2018-04-02:
Try going through this :--

https://github.com/attify/firmware-analysis-toolkit/issues/15

extremecoders-re, 2018-04-02:
@jkbecker You can follow the write-up on our blog on emulating the Netgear wnap320 firmware which also includes steps for setting up fat.

https://blog.attify.com/getting-started-with-firmware-emulation/

seebog, 2018-12-28:
@jkbecker I also encountered the same problem as you, how can you solve it?

BeetleChunks, 2019-06-25:
@seebog @jkbecker I had this issue and figured out how to fix it. Look at the error message and note the line starting with "args: ". This is being executed by pexpect and causing the error, as the response from the command is not what fat.py is expecting.

Copy that line out and run it manually:
```
/root/device-simulator/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py -b netgear -sql 127.0.0.1 -np -nk /root/Desktop/WNAP320 Firmware Version 2.0.3.zip images
```

In my case I saw the issue was related to an unmet dependency "psycopg2" which lead to the dependency of "libpq-dev". Below are the commands to fix this issue if they are the same as mine, if not that is why I provided a detailed description of how I found the issue on my machine.

Command that fixed it for me:
```
sudo apt install libpq-dev
pip install psycopg2
```

Hope this helps.

adi0x90, 2026-09-25:
The extractor exited while `fat.py` was still waiting for `Database Image ID:`. That left you with a `pexpect` traceback without a useful explanation of the extraction failure.

FAT 2 no longer reads image IDs from `extractor.py` prompts. It uses `fat identify` and project-based `fat extract`, with separate steps for [extraction, analysis, and emulation](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime).

Closing this since that wrapper is gone. If WNAP320 still fails in v2, please open an issue with the exact firmware version and extraction output. Thanks for checking more than one image and including the trace.

closed 2026-09-25 by adi0x90