attify/firmware-analysis-toolkit #17: Network interface not working and no IP assigned.

r00tb3, 2018-04-03:
I'm getting the following error:--

```
     __           _   
                              / _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : /home/r00tb3/Desktop/dir-816/firmware/AP699E8C.CW125A-5-DLINK-R1B011D81870(0519085924)(1).img
[?] Enter the brand of the firmware : dlink
[+] Now going to extract the firmware. Hold on..
[+] Firmware : /home/r00tb3/Desktop/dir-816/firmware/AP699E8C.CW125A-5-DLINK-R1B011D81870(0519085924)(1).img
[+] Brand : dlink
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : mipsel
[+] Storing filesystem in database
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : []
[+] Running the firmware finally
[+] command line : sudo /home/r00tb3/re-tools/firmware-analysis-toolkit/firmadyne/scratch/1/run.sh
[*] Press ENTER to run the firmware...^CTraceback (most recent call last):

```

Why isn't it enabling any network interface?

How do I get this working and get and IP assigned to it?

Waiting for replies!!

extremecoders-re, 2018-04-03:
Unfortunately, not all firmware image work out of the box with firmadyne. Looks like the inferNetwork.sh script failed to detect any interface. Try increasing the timeout value in [inferNetwork.sh](https://github.com/firmadyne/firmadyne/blob/master/scripts/inferNetwork.sh) which is a part of upstream firmadyne.

[Steps here](https://github.com/firmadyne/firmadyne#runsh-is-not-generated)

r00tb3, 2018-04-03:
@extremecoders-re The problem is that Init crashes:--

```
nvram_get_buf: Unable to open key: /firmadyne/libnvram/RESET_TAG!
nvram_get_buf: WebInit
sem_get: Key: 4110002a
sem_get: Key: 4110002a
nvram_get_buf: Unable to open key: /firmadyne/libnvram/WebInit!
nvram_clear: Clearing NVRAM...
sem_lock: Already initialized!
```

I think I'll have to recompile the libnvram but I don't know how to do it.

Pls, help me with this.

Thanks for the reply!!

extremecoders-re, 2018-04-04:
You can include additional entries by modifying [config.h](https://github.com/firmadyne/libnvram/blob/master/config.h).

extremecoders-re, 2018-04-04:
Check the error messages `nvram_get_buf: Unable to open key:`

DerZc, 2019-01-09:
@r00tb3 Have you solved this problem? How did you do?

blakroot, 2019-02-24:
Is anyone able to solve this problem?

r00tb3, 2019-02-25:
No i haven't solved the problem yet.

anindya14, 2021-04-29:
did you get the solution of it @r00tb3 ?

mark0519, 2022-03-21:
Is anyone able to solve this problem now?

devika6001, 2023-11-14:
@mark0519 are you able to solve this problem?

sdahiasdba, 2024-04-02:
@mark0519 @extremecoders-re  @anindya14  @r00tb3  @DerZc  @mark0519 Anyone solve this problem?!!!

adi0x90, 2026-09-25:
The empty interface list doesn't tell me whether this DIR-816 image booted far enough to configure its network.

FAT 2's native `qemu-direct` system mode uses host-port forwarding instead of the old network-inference script. The guest still needs to boot and start its services, and DIR-816 isn't covered by the successful extraction tests listed here.

Please add the image hash, backend, and extraction/preflight/serial output from a v2 run. That'll show where this particular image first gets stuck.

adi0x90, 2026-09-26:
I found the DIR-816 filename in your log, but have not found a working download for that specific build: `AP699E8C.CW125A-5-DLINK-R1B011D81870(0519085924)(1).img`.

Could you share a current download link and the file's SHA-256? [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) improves extraction, but I have not tested this exact image or resolved its init/network failure.

For a retry, please use [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1), a fresh project, and include the backend and new serial log.