attify/firmware-analysis-toolkit #18: Architecture : ./images/1.tar.gz: Cannot open: No such file or directory

kevins1022, 2018-04-20:
when i run fat.py

i meet this error.

```
root@ubuntu:~/firmware-analysis-toolkit/firmadyne# sudo python fat.py

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme

[?] Enter the name or absolute path of the firmware you want to analyse : wnap320.zip
[?] Enter the brand of the firmware : fff
[+] Now going to extract the firmware. Hold on..
[+] Firmware : wnap320.zip
[+] Brand : fff
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : ./images/1.tar.gz: Cannot open: No such file or directory
Traceback (most recent call last):
  File "fat.py", line 122, in <module>
    main()
  File "fat.py", line 114, in main
    arch = identify_arch(image_id)
  File "fat.py", line 62, in identify_arch
    child.expect("Password for user firmadyne: ")
  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1418, in expect
    timeout, searchwindowsize)
  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1433, in expect_list
    timeout, searchwindowsize)
  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1521, in expect_loop
    raise EOF(str(err) + '\n' + str(self))
pexpect.EOF: End Of File (EOF). Exception style platform.
<pexpect.spawn object at 0x7f31750b26d0>
version: 3.1
command: /root/firmware-analysis-toolkit/firmadyne/scripts/getArch.sh
args: ['/root/firmware-analysis-toolkit/firmadyne/scripts/getArch.sh', './images/1.tar.gz']
searcher: <pexpect.searcher_re object at 0x7f31750b2710>
buffer (last 100 chars): ''
before (last 100 chars): 'tar: Error is not recoverable: exiting now\r\n'
after: <class 'pexpect.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 27475
child_fd: 3
closed: False
timeout: 30
delimiter: <class 'pexpect.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
```

kevins1022, 2018-04-20:
Reinstalled once   and successful..

closed 2018-04-20 by kevins1022

icommitfelonies, 2018-12-09:
reinstalled what?

aquarius80lvl, 2019-04-13:
same issue: https://github.com/attify/firmware-analysis-toolkit/issues/27