attify/firmware-analysis-toolkit #21: Netgear firmware unable to simulate in the firmadyne

susantaroy2002, 2018-07-03:
Unable to simulate Netgear firmware, not if there any thing wrong on .chk  file extension. 

Here I am sharing some details about my blocker, 
1. Tried on 2 netgear firmware version - "D8500-V1.0.3.39_1.0.1.chk" and "R7000-V1.0.9.28_10.2.32.chk"
2. attaching the error screenshot. 

Please let me know if you need any further details to investigate this issue.

[image]
[image]
[image]

extremecoders-re, 2018-07-03:
You are using an old version of fat. Consider updating it.

As for the actual problem, The file "run.sh" isn't generated, thats why its failing. You can have a look here how to proceed in such cases.

https://github.com/firmadyne/firmadyne#runsh-is-not-generated

adi0x90, 2018-07-09:
It also looks like there's a conflict with one of the previous runs. 

Can you run the `reset.py` once and try it again.

susantaroy2002, 2018-08-02:
@adi0x90 Hi Aditya, as per your direction I have tried below options - 

First ran the `download.sh` and then `reset.py` , however still the same issue. 

[image]

adi0x90, 2018-08-03:
@susantaroy2002 : Can you share the firmware binaries?

susantaroy2002, 2018-08-03:
@adi0x90 Thanks for your quick response, I am attaching the firmware binary. 

Here is the download link - https://we.tl/GGcBtcUQ83

twisted-fun, 2019-03-13:
Updates on this one? I am trying to emulate Netgear firmware - R7000 and D8500 (latest versions). In both, I am unable to get any network interfaces. After running run.sh, I get kernel panic.

```
> sudo ./fat.py  

                               __           _   
                              / _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : R7000-V1.0.9.64_10.2.64.zip
[?] Enter the brand of the firmware : Netgear
[+] Now going to extract the firmware. Hold on..
[+] Firmware : R7000-V1.0.9.64_10.2.64.zip
[+] Brand : Netgear
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : armel
[+] Storing filesystem in database
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : []
[+] Running the firmware finally
[+] command line : sudo /home/oit/firmadyne/scratch/1/run.sh
[*] Press ENTER to run the firmware...
Starting firmware emulation... use Ctrl-a + x to exit
attify123
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.17+ (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:05:21 UTC 2016
[    0.000000] CPU: ARMv7 Processor [412fc0f1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
...
many kernel messages
kernel panic afterwards
```

Tried resetting also.. same issue again

adi0x90, 2026-09-25:
There's progress on R7000 extraction: the v2 test log shows a `.chk` image unpacked into a root filesystem with 2,087 files. Thanks for including both model and firmware versions in the original report.

I wouldn't mark emulation fixed from that result. There's no matching boot or service-access result here, D8500 hasn't been covered by these tests, and the R7000 test image still needs to be matched to your version/hash.

Please add the image version/hash, backend, and serial log for each model tried with the [v2 workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime). Extraction is working for the tested sample; boot and service access still need checking.

adi0x90, 2026-09-26:
I now have extraction results for all three releases named in this thread: D8500 `1.0.3.39_1.0.1`, R7000 `1.0.9.28_10.2.32`, and the later R7000 `1.0.9.64_10.2.64`. After [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108), each native extraction recovered a rootfs whose regular files and symlink targets matched independent reference extraction. These extraction checks used case-sensitive APFS.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'm keeping this open because the reported failure is emulation. Please share the firmware version, `fat --version` output, backend, and serial log from a fresh attempt if it still stops before the web interface becomes available. The extraction checks have not established a successful boot.