attify/firmware-analysis-toolkit #22: Failed on Architecture Step

jocko96b, 2018-10-03:
I am beating my head on this, trying to get it to run for the first time. I followed the procedures here: https://securityonline.info/firmware-analysis-toolkit/. The firmadify database setup was missing which I completed using steps in the firmadify git homepage.

When running, I get the following:

[?] Enter the name or absolute path of the firmware you want to analyse : /firmware/DIR645A1_FW105B01.bin
[?] Enter the brand of the firmware : dlink
[+] Now going to extract the firmware. Hold on..
[+] Firmware : /firmware/DIR645A1_FW105B01.bin
[+] Brand : dlink
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : Could not find 'firmadyne.config'!
Traceback (most recent call last):
File "./fat.py", line 121, in 
main()
File "./fat.py", line 113, in main
arch = identify_arch(image_id)
File "./fat.py", line 61, in identify_arch
child.expect("Password for user firmadyne: ")
File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
timeout, searchwindowsize, async_)
File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
return exp.expect_loop(timeout)
File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
return self.eof(e)
File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f86ceae1650>
command: /tools/firmadyne//scripts/getArch.sh
args: ['/tools/firmadyne//scripts/getArch.sh', './images/1.tar.gz']
buffer (last 100 chars): ''

after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 18582
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
0: re.compile('Password for user firmadyne: ')

Here are my settings from fat.py:

#Configurations - change this according to your system
firmadyne_path = "/tools/firmadyne"
binwalk_path = "/tools/binwalk"
root_pass = "root"
firmadyne_pass = "firmadyne"

And my settings from firmadyne.conf (which is located in /tools/firmadyne/)

#uncomment and specify full path to FIRMADYNE repository
FIRMWARE_DIR=/tools/firmadyne

Thank you in advance for your help!

extremecoders-re, 2018-10-03:
I think the paths are incorrect. 

```
#Configurations - change this according to your system
firmadyne_path = "/tools/firmadyne"
binwalk_path = "/tools/binwalk"
```

Are you sure there a `tools` directory directly under root `/` ? 

Usually it's under the home directory.
```
firmadyne_path = "~/tools/firmadyne"
```

jocko96b, 2018-10-03:
No, I created the tools directory in the root where I started dumping everything into.  
Definitely not in the home directory.

extremecoders-re, 2018-10-03:
Can you check if you running fat.py from the `/tools/firmadyne` directory?
If not just copy fat.py and reset.py to that dir and try again.

jocko96b, 2018-10-03:
That got it!  

The steps I was following from here: https://securityonline.info/firmware-analysis-toolkit/ had me move the two .py files up a level for whatever reason.  

All is well - thank you so much for the help!

closed 2018-10-03 by jocko96b