attify/firmware-analysis-toolkit #23: Unable to execute the execute extractor.py

Minyanaing, 2018-11-09:
When I run the fat.py, the program start and then I put the firmware and the brand. Then the error occurred as described below.
"""
Traceback (most recent call last):
File "fat.py", line 122, in
main()
File "fat.py", line 109, in main
image_id = run_extractor(firm_name, firm_brand)
File "fat.py", line 47, in run_extractor
child = pexpect.spawn(extractor_cmd, timeout=None)
File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 204, in init
self._spawn(command, args, preexec_fn, dimensions)
File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 276, in _spawn
'executable: %s.' % self.command)
pexpect.exceptions.ExceptionPexpect: The command was not found or was not executable: /home/firmadyne/sources/extractor/extractor.py.
"""

extremecoders-re, 2018-11-09:
From the error message, looks like firmadyne (on which fat is based) isn't setup properly. It's failing on the line.
```
child = pexpect.spawn(extractor_cmd, timeout=None)
```

adi0x90, 2026-09-25:
The old script was looking for `extractor.py` at the configured Firmadyne path and couldn't find an executable there.

FAT 2's `fat extract` doesn't call that script, so you no longer need to edit its location in `fat.py`. Follow the [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md), then run `fat doctor` to check the external tools.

Closing this since the missing-script dependency is gone. If v2 reports a missing tool, please include the command and `fat doctor` output in a new issue.

closed 2026-09-25 by adi0x90