attify/firmware-analysis-toolkit #24: The command was not found or was not executable: sudo

DwijadasDey, 2018-11-13:
Hi
I am trying to run fat.py using following command only to be informed that sudo command is not executable.

`$ python /usr/local/bin/fat.py openwrt-ar71xx-generic-mr600-squashfs-factory.bin`

[?] Enter the name or absolute path of the firmware you want to analyse : openwrt-ar71xx-generic-mr600-squashfs-factory.bin
[?] Enter the brand of the firmware : op
[+] Now going to extract the firmware. Hold on..
[+] Firmware : openwrt-ar71xx-generic-mr600-squashfs-factory.bin
[+] Brand : op
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : mipseb
[+] Storing filesystem in database
[!] Filesystem already exists
[+] Building QEMU disk image
Traceback (most recent call last):
  File "/usr/local/bin/fat.py", line 122, in <module>
    main()
  File "/usr/local/bin/fat.py", line 116, in main
    make_image(arch, image_id)
  File "/usr/local/bin/fat.py", line 80, in make_image
    child = pexpect.spawn(makeimage_cmd)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 204, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 276, in _spawn
    'executable: %s.' % self.command)
pexpect.exceptions.ExceptionPexpect: The command was not found or was not executable: sudo.

Is it due to i have copied fat.py in the location /usr/local/bin ? Any inputs will be highly appreciated.
Regards

DwijadasDey, 2018-11-13:
The sudo package was not installed. After installing it FAT is working correctly.

extremecoders-re, 2018-11-13:
Hi, good to know that you got it working. If your problem is solved you can close the issue.