attify/firmware-analysis-toolkit #25: Architecture : Could not find 'firmadyne.config'!

DwijadasDey, 2018-11-14:
Hi
I have installed firmware analysis toolkit along with all dependencies in Ubuntu 16.04. The directory structure is like below.

```
/fat
     /firmadyne
     /firmwalker
     /firmware-analysis-toolkit
     /firmware-mod-kit
```

Changes the firmadyne path in `/fat/firmware-analysis-toolkit/fat.py`
```
....
FIRMWARE_DIR=/fat/firmadyne/

# specify full paths to other directories
BINARY_DIR=${FIRMWARE_DIR}/binaries/
TARBALL_DIR=${FIRMWARE_DIR}/images/
SCRATCH_DIR=${FIRMWARE_DIR}/scratch/
SCRIPT_DIR=${FIRMWARE_DIR}/scripts/
....
```

firmadyne.config is in `/fat/firmadyne/firmadyne.config` and `FIRMWARE_DIR` points to `/fat/firmadyne`

Executed the fat.py using following command:

`$  python /fat/firmware-analysis-toolkit/fat.py topo.bin`

But the above command produces **Could not find 'firmadyne.config** with following output.

```
[?] Enter the name or absolute path of the firmware you want to analyse : topo.bin
[?] Enter the brand of the firmware : lk
[+] Now going to extract the firmware. Hold on..
[+] Firmware : topo.bin
[+] Brand : lk
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : Could not find 'firmadyne.config'!
Traceback (most recent call last):
  File "/fat/firmware-analysis-toolkit/fat.py", line 122, in <module>
    main()
  File "/fat/firmware-analysis-toolkit/fat.py", line 114, in main
    arch = identify_arch(image_id)
  File "/fat/firmware-analysis-toolkit/fat.py", line 62, in identify_arch
    child.expect("Password for user firmadyne: ")
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f6ed2ad5650>
command: /fat/firmadyne/scripts/getArch.sh
args: ['/fat/firmadyne/scripts/getArch.sh', './images/1.tar.gz']
buffer (last 100 chars): ''

after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 889
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('Password for user firmadyne: ')
```
What is wrong with the above setings ? Does all files needs to be in the same folder for firmadyne
and firmware-analysis-toolkit ?

DwijadasDey, 2018-11-14:
Copied fat.py and reset.py to /fat/firmadyne command and issuing the command from there.

```
cd /fat/firmadyne
python fat.py latlon.bin
```

The previous error has gone away but another pops up.

```
[?] Enter the name or absolute path of the firmware you want to analyse : latlon.bin
[?] Enter the brand of the firmware : qa
[+] Now going to extract the firmware. Hold on..
[+] Firmware : latlon.bin
[+] Brand : qa
[+] Database image ID : 2
[+] Identifying architecture
[+] Architecture : ./images/2.tar.gz: Cannot open: No such file or directory
Traceback (most recent call last):
  File "fat.py", line 122, in <module>
    main()
  File "fat.py", line 114, in main
    arch = identify_arch(image_id)
  File "fat.py", line 62, in identify_arch
    child.expect("Password for user firmadyne: ")
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f875b7da650>
command: /fat/firmadyne/scripts/getArch.sh
args: ['/fat/firmadyne/scripts/getArch.sh', './images/2.tar.gz']
buffer (last 100 chars): ''
before (last 100 chars): 'mages/2.tar.gz: Cannot open: No such file or directory\r\ntar: Error is not recoverable: exiting now\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 328
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('Password for user firmadyne: ')
```

extremecoders-re, 2018-11-14:
You have to edit `fat.py` and `reset.py` and provide the correct root password if not already done so.

https://github.com/attify/firmware-analysis-toolkit/blob/fa5d8f4639dff045c80519d693630debd17d0439/fat.py#L13-L14

DwijadasDey, 2018-11-15:
Hi
Thanks for the reply !

I have the firmadyne user in the postgresql database with the password 'firmadyne'

I have also executed the following query to make sure the user 'firmadyne' exists in the database.

```
postgres=# sudo -u postgres createuser firmadyne
postgres=# alter user firmadyne with encrypted password 'firmadyne'
postgres=# grant all privileges on database firmadyne to firmadyne
```

```
postgres-# \l
                              List of databases
   Name    |   Owner   | Encoding | Collate |  Ctype  |   Access privileges
-----------+-----------+----------+---------+---------+-----------------------
 firmadyne | firmadyne | UTF8     | C.UTF-8 | C.UTF-8 |
 firmware  | firmadyne | UTF8     | C.UTF-8 | C.UTF-8 |
```
But the error still persists.

extremecoders-re, 2018-11-15:
Hi, looks like you're missing a step somewhere. 
You can refer to this blog post on how to setup FAT https://blog.attify.com/getting-started-with-firmware-emulation/

DwijadasDey, 2018-11-16:
Started afresh and Recompiled but looks its producing much before in the database image section.

```
[?] Enter the name or absolute path of the firmware you want to analyse : tdpack5.bin
[?] Enter the brand of the firmware : tdpack5
[+] Now going to extract the firmware. Hold on..
[+] Firmware : tdpack5.bin
[+] Brand : tdpack5
Traceback (most recent call last):
  File "fat.py", line 122, in <module>
    main()
  File "fat.py", line 109, in main
    image_id = run_extractor(firm_name, firm_brand)
  File "fat.py", line 48, in run_extractor
    child.expect("Database Image ID: ")
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f0bc003b650>
command: /fat/firmware-analysis-toolkit/firmadyne//sources/extractor/extractor.py
args: ['/fat/firmware-analysis-toolkit/firmadyne//sources/extractor/extractor.py', '-b', 'tdpack5', '-sql', '127.0.0.1', '-np', '-nk', 'tdpack5.bin', 'images']
buffer (last 100 chars): ''
before (last 100 chars): 'bound method ExtractionItem.__del__ of <__main__.ExtractionItem object at 0x7f9da5c53f10>> ignored\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 80
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('Database Image ID: ')
```
Postgresql is running with user firmadyne and database firmware

DwijadasDey, 2018-11-16:
Hi
I was testing with a wrong bin file from https://www.ncl.ucar.edu/Applications/Data/ and when i changed the binary file from http://files.dlink.com.au/products/DIR-850L/REV_B/Firmware/Firmware_v2.20b03/ the FAT works producing outputs but at the end the kernel got panic and system needs a restart.

```
[    1.872000] No filesystem could mount root, tried:  ext3 ext2 ext4 cramfs squashfs vfat iso9660 romfs udf
[    1.880000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(8,1)
```
root_pass is defined in fat.py

extremecoders-re, 2018-11-17:
Hi, just checked on the `DIR850LB1_FW210WWb03.bin` file. Its working well.

```
$ ./fat.py DIR850LB1_FW210WWb03.bin 

                               __           _   
                              / _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : DIR850LB1_FW210WWb03.bin
[?] Enter the brand of the firmware : dlink
[+] Now going to extract the firmware. Hold on..
[+] Firmware : DIR850LB1_FW210WWb03.bin
[+] Brand : dlink
[+] Database image ID : 2
[+] Identifying architecture
[+] Architecture : mipseb
[+] Storing filesystem in database
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
[+] Running the firmware finally
[+] command line : sudo /home/ec/tools/firmadyne/scratch/2/run.sh
[*] Press ENTER to run the firmware...
```
The web server is reachable as well.
[image]

DwijadasDey, 2018-11-17:
Hi
I am running FAT inside docker. Want to know if the `No filesystem could mount root`  as mentioned in my last post is okay. Also i don't see any network interfaces as well during initialization.

```
 [?] Enter the name or absolute path of the firmware you want to analyse : DIR850LB1_FW210WWb03.bin
[?] Enter the brand of the firmware : dlink
[+] Now going to extract the firmware. Hold on..
[+] Firmware : DIR850LB1_FW210WWb03.bin
[+] Brand : dlink
[+] Database image ID : 2
[+] Identifying architecture
[+] Architecture : mipseb
[+] Storing filesystem in database
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
**[+] Network interfaces : []**
[+] Running the firmware finally
[+] command line : sudo /fat/firmware-analysis-toolkit/firmadyne//scratch/2/run.sh
[*] Press ENTER to run the firmware...
```

extremecoders-re, 2018-11-18:
No, it shouldn't happen. Same for the network interface.

I've tested the image on Ubuntu 18.04. Sometimes using a different version of qemu does magically makes such issue disappear. Firmware emulation is quite an experimental thing and you are bound to run into such problems every now and then.

gambr0, 2019-10-24:
Hi
I had the same error as yours and my directory structure was like yours. But after I tried following steps, it seems to work well now:
1. move /firmadyne into /firmware-analysis-toolkit
2. navigate to the Firmadyne folder and execute "sudo ./download.sh"  (seems like I missed this step when setting up Firmadyne)
3.  move fat.py and reset.py into Firmadyne folder
4. modify the firmadyne_path (in fat.py) and FIRMWARE_DIR (in firmadyne.config) to the current path of Firmadyne.

adi0x90, 2026-09-25:
The directory layout helps explain where this failed. The old `getArch.sh` step was looking for `firmadyne.config` while processing the extracted archive.

FAT 2 uses `fat identify --file ./firmware.bin` for identification and doesn't need that config file for this step. The [project workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) covers what comes next.

Closing this since that script and config dependency have been replaced. If identification still comes back incomplete in v2, send the model/version and output in a new issue so I can look at what it missed.

closed 2026-09-25 by adi0x90