attify/firmware-analysis-toolkit #27: ./images/1.tar.gz: Cannot open: No such file or directory

icommitfelonies, 2018-12-09:
I keep running into the same issue, having installed your toolkit and all dependencies a couple of different ways now. The only other issue regarding this error was closed in the same day with the resolution being to reinstall, which hasn't worked.

> [+] Database image ID : 1
> [+] Identifying architecture
> [+] Architecture : ./images/1.tar.gz: Cannot open: No such file or directory
> Traceback (most recent call last):
>   File "fat.py", line 122, in <module>
>     main()
>   File "fat.py", line 114, in main
>     arch = identify_arch(image_id)        
>   File "fat.py", line 62, in identify_arch
>     child.expect("Password for user firmadyne: ")    
>   File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
>     timeout, searchwindowsize, async_)
>   File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
>     return exp.expect_loop(timeout)
>   File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
>     return self.eof(e)
>   File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
>     raise EOF(msg)
> pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
> <pexpect.pty_spawn.spawn object at 0x7f5bbcb43110>
> command: /home/labuser/firmadyne/scripts/getArch.sh
> args: ['/home/labuser/firmadyne/scripts/getArch.sh', './images/1.tar.gz']
> buffer (last 100 chars): ''
> before (last 100 chars): 'mages/1.tar.gz: Cannot open: No such file or directory\r\ntar: Error is not recoverable: exiting now\r\n'
> after: <class 'pexpect.exceptions.EOF'>
> match: None
> match_index: None
> exitstatus: 1
> flag_eof: True
> pid: 8064
> child_fd: 5
> closed: False
> timeout: 30
> delimiter: <class 'pexpect.exceptions.EOF'>
> logfile: None
> logfile_read: None
> logfile_send: None
> maxread: 2000
> ignorecase: False
> searchwindowsize: None
> delaybeforesend: 0.05
> delayafterclose: 0.1
> delayafterterminate: 0.1
> searcher: searcher_re:
>     0: re.compile('Password for user firmadyne: ')

What am I doing wrong?

extremecoders-re, 2018-12-09:
Hi, as I see from the log above the error is likely because the firmware binary was not extracted properly. Can you check if `1.tar.gz` is present within the `images` directory?

If the directory is indeed empty, can you run `binwalk` and check if it does detect any filesystem?

icommitfelonies, 2018-12-09:
That's what I thought was happening... binwalk does not seem to identify a fileysystem unless I use the -I (invalid) flag. It identifies EFS2 (with -I) in numerous places throughout the firmware. I've attempted extraction directly with binwalk several times with no luck.

Fairly new to IoT and Firmware testing, so I appreciate your response.

extremecoders-re, 2018-12-09:
Looks interesting! Can you share the firmware binary.

aquarius80lvl, 2019-04-13:
Here`s another example which has a similar issue: https://drive.google.com/open?id=1FSf9Tc8TI3tE1dC9Ylw_dTCTA_54rGRT
The tar.gz is not being created in the images/ , however, binwalk -e is able to extract the fs

syhxsqqZM, 2019-11-08:
when I run :extractor.py -b netgear -sql 127.0.0.1 -np -nk "wnap320.zip" images ,I found that error is found in magic. So I reinstalled the python-magic library throuth the method that download from github  and install by using 'python setup.py install'.finally,I solved the problem.

adi0x90, 2026-09-25:
Sorry reinstalling didn't help. The traceback shows `fat.py` trying to identify the architecture from an archive it couldn't find.

FAT 2 uses projects for extraction and analysis, so it no longer needs that `./images/1.tar.gz` step. The [current workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) starts with `fat new`, then `fat extract` and the later analysis/preflight steps.

Closing this since the old script has been replaced. I can't confirm from the results here that your firmware works in v2. If it still fails, please open a new issue with the model/version, command, and output.

closed 2026-09-25 by adi0x90