attify/firmware-analysis-toolkit #28: When building the qemu image, there will be a timeout

DerZc, 2019-01-09:
Hi,
When I building the qemu disk image, there will be a timeout. 
[image]
How can I do to fix it?
Thank you

extremecoders-re, 2019-01-09:
Can you share the firmware binary?

DerZc, 2019-01-10:
Hi,
@extremecoders-re The binary I used is getting from this link [https://support.dlink.com/ProductInfo.aspx?m=DIR-880L](https://support.dlink.com/ProductInfo.aspx?m=DIR-880L) ,but this problem only occurs at certain times.
I also meet the problem that I can't get the IP address, I read the log file and find that he is not one of the following three situations.
[run.sh is not generated](https://github.com/firmadyne/firmadyne#runsh-is-not-generated)
When I read the log file I find your tool can find the RAM and ROM address from the firmware binary. Can you tell me how you did it?
Thank you very much!!

Silipwn, 2019-01-11:
Stuck on a same issue @extremecoders-re :|

extremecoders-re, 2019-01-11:
@DerZc I checked the firmware binary. It's for an ARM device, however as per your screenshot its wrongly identified as mipsel. So may be you're using an older version of firmadyne. Here its correctly being identifed as arm.

The `proceed anyway (y, N)` message you're getting is from these lines in [makeImage.sh](https://github.com/firmadyne/firmadyne/blob/master/scripts/makeImage.sh#L77-L79)

```
echo "----Creating Filesystem----"
mkfs.ext2 "${DEVICE}"
sync
```

You can temporarily work around the error by forcing it to create the filesystem rather than prompting for user input. `mkfs.ext2 -F "${DEVICE}"`

Finally, this particular firmware binary DOES have issues with the recent versions of qemu. In my tests here, using an older version, I was able to boot to a terminal.

```
qemu-system-arm -version  
QEMU emulator version 2.0.0 (Debian 2.0.0+dfsg-2ubuntu1.30), Copyright (c) 2003-2008 Fabrice Bellard
```
---
> I also meet the problem that I can't get the IP address,

Using an older qemu may work
```
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
[+] Running the firmware finally
[+] command line : sudo /home/oit/tools/firmadyne-new/scratch/1/run.sh
```
---

> When I read the log file I find your tool can find the RAM and ROM address from the firmware binary. Can you tell me how you did it?

This is done my inspecting the kernel boot log which is the same as you see on screen. This is a part of upstream firmadyne.

---
@Silipwn Its looks to be waiting on the `mkfs.ext2 "${DEVICE}"` call.

Overall, I would say this particular firmware does not play well with firmadyne out of the box. You may need to change some of the scripts in order to boot it successfully.

Silipwn, 2019-01-12:
@extremecoders-re Thanks for the help :+1: :)

DerZc, 2019-01-12:
@extremecoders-re Sorry to bother you again. I use a older QEMU but I also can not get the IP
[image]
Can you help me, thank you very much

DerZc, 2019-01-13:
@extremecoders-re I get the IP address but I can not ping it
[image]

DerZc, 2019-01-13:
[image]
and he is stuck in here

DerZc, 2019-01-14:
@extremecoders-re Thank you for your help.
I get the IP address but I can not ping it. 
[image]
Can you tell me how can I solve this question?

extremecoders-re, 2019-01-14:
@DerZc Ping doesn't always work. For testing the firmware, all you need is to be able to access the web sever. To this end,  I would recommend to edit the `run.sh` script and change from `tap` to user-mode networking, using the `-redir tcp` options. Then you may try to curl/wget/nc to the IP both from your host and within the qemu console.

As you can see, firmadyne doesn't work in all cases out of the box. Emulating a device with qemu will always be experimental and not guaranteed to work every-time unless you are willing to put a lot of effort which includes editing the firmware itself to make it compatible.

In many cases, you can use qemu user mode emulation to run just the webserver for testing. I would recommend this approach as it's a lot easier.

adi0x90, 2026-09-25:
This timeout came from the old `fat.py` workflow for building a Firmadyne disk image. Thanks for including the DIR-880L reference and noting that it happened intermittently.

FAT 2's native `qemu-direct` system mode prepares and launches its own image, without that `makeImage.sh` step. The [v2 project workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) covers the new setup.

Closing this since the old image-building step has been replaced on that path. I still don't have a confirmed DIR-880L boot result here. If preparation stalls in v2, please send the backend, command, and last output in a new issue.

closed 2026-09-25 by adi0x90