attify/firmware-analysis-toolkit #30: Clear-text password in script

alessandroZelli, 2019-03-01:
Both fat.py and reset.py require us to set a clear text password in the source code. 
Is there a workaround for that?

extremecoders-re, 2019-03-01:
Ideally, this tool should be run in a Virtual Machine so that specifying the password is not an issue.
If not you can use an older version which asks for the password every-time rather than automating it. 

https://github.com/attify/firmware-analysis-toolkit/tree/dc64ccfeba8c549cd78c67c806472f956f221720

alessandroZelli, 2019-03-01:
Ok, will do that. I'll try to come up with a secure way to run it on a main OS as soon as I have some time.

extremecoders-re, 2019-03-02:
The sudoers password. 

However the reason for recommending a VM is not just this. During emulation the script will set up a TAP interface for the emulated firmware image. So it does happen sometime that due to some error or otherwise this interface is not removed after emulation is done. 
In that case you'll be left to manually cleanup and the networking of your main OS may be affected. Hence its better to use a VM.

adi0x90, 2026-09-25:
You shouldn't have had to save a sudo password in a script or config file. Thanks for calling that out.

FAT 2 no longer uses the old `fat.py`/`reset.py` scripts or the `fat.config`/`sudo_password` setup. Installation defaults to `~/.local`. If you use `--install-system-deps`, the installer calls `sudo` normally; it doesn't ask you to store the password in FAT.

Closing this since the password-storage code is gone. The [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md) covers the new setup. If you still have a v1 checkout, remove any password you saved in that copy too.

closed 2026-09-25 by adi0x90