attify/firmware-analysis-toolkit #32: what is AttifyOS's password?

fIappy, 2019-06-25:
what is AttifyOS's password?

extremecoders-re, 2019-06-25:
This one?
https://github.com/adi0x90/attifyos

It's `attify123`

fIappy, 2019-06-25:
> This one?
> https://github.com/adi0x90/attifyos
> 
> It's `attify123`

Thank you very much! I means this one  ,http://tinyurl.com/attifyos, which is mentioned by "In case you are on Kali and are facing issues with emulation, it is recommended to use the AttifyOS Pre-Release VM downloadable from here, or alternatively you could do the above mentioned."

fIappy, 2019-06-25:
> Thank you very much! I means this one ,http://tinyurl.com/attifyos, which is mentioned by "In case you are on Kali and are facing issues with emulation, it is recommended to use the AttifyOS [...]

I got it! It's password@123

closed 2019-06-25 by fIappy