attify/firmware-analysis-toolkit #33: Error running fat.py with openwrt firmware

fIappy, 2019-08-18:
this is my problem:
> ./fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : homerouter.bin
Enter the brand of the firmware : asd
homerouter.bin
Now going to extract the firmware. Hold on..
/home/oit/tools/fat//sources/extractor/extractor.py -b asd -sql 127.0.0.1 -np -nk "homerouter.bin" images 
test
The database ID is 5
Getting image type
Password for user firmadyne: 
Found image type of  mipseb
Putting information to database
Traceback (most recent call last):
  File "/home/oit/tools/fat//scripts/tar2db.py", line 100, in <module>
    main()
  File "/home/oit/tools/fat//scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "/home/oit/tools/fat//scripts/tar2db.py", line 77, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "/home/oit/tools/fat//scripts/tar2db.py", line 57, in insertObjectToImage
    for x in files2oids])
psycopg2.IntegrityError: duplicate key value violates unique constraint "object_to_image_oid_iid_filename_key"
DETAIL:  Key (oid, iid, filename)=(3800, 5, /etc/uci-defaults/04_led_migration) already exists.

Already done earlier
Creating Image
Executing command 

sudo /home/oit/tools/fat//scripts/makeImage.sh 5
Password for user firmadyne: 
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0x0fd25194.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
Building a new DOS disklabel with disk identifier 0x80288bfb.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.9 (4-Feb-2014)
Please check the makeImage function
Everything is done for the image id 5
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 8417
Querying database for architecture... mipseb
Running firmware 5: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 
sudo: /home/oit/tools/fat//scratch/5/run.sh: command not found
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 109, in main
    final_run(image_id)
  File "./fat.py", line 89, in final_run
    print subprocess.check_output(final_run_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /home/oit/tools/fat//scratch/5/run.sh' returned non-zero exit status 1

thanks

fIappy, 2019-08-18:
the firemware can be downloaded in https://adworld.xctf.org.cn/media/uploads/task/3dfb553b36f5471991c774569a929ef5.zip

adi0x90, 2026-09-25:
There are two errors in this output: a duplicate key in Firmadyne's database, followed by an attempt to launch a `scratch/.../run.sh` file that wasn't there. I wouldn't call this an OpenWrt compatibility failure from that output alone.

FAT 2 replaces the old `fat.py` database and script setup with the [project workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime).

Closing this since the old setup has been retired. If the same OpenWrt build fails in v2, please send the build version, command, and output in a new issue so I can see whether it stops at extraction, boot, or service startup.

closed 2026-09-25 by adi0x90