attify/firmware-analysis-toolkit #34: can't run fat to ananlysis firmware

owenbill, 2019-09-27:
Hi All,
when i running sudo ./fat.sh then i got following error 
[+] Architecture : line 7: ./firmadyne.config: Permission denied
Traceback (most recent call last):
  File "./fat.py", line 122, in <module>
    main()
  File "./fat.py", line 114, in main
    arch = identify_arch(image_id)        
  File "./fat.py", line 62, in identify_arch
    child.expect("Password for user firmadyne: ")    
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f4a16241610>
command: /home/bill/firmware-analysis-toolkit/firmadyne/scripts/getArch.sh
args: ['/home/bill/firmware-analysis-toolkit/firmadyne/scripts/getArch.sh', './images/1.tar.gz']
buffer (last 100 chars): ''

after: <class 'pexpect.exceptions.EOF'>
and i use user permission to install so how to check the issue? Thanks

kismp123, 2020-04-02:
I think you should run extrator.py. as you wrote, I also met this problem.
I don't remember how I solve. anyway you should run extractor.py and see logs.

adi0x90, 2026-09-25:
The first error is `Permission denied` when `getArch.sh` reads `firmadyne.config`. The `pexpect` EOF comes after that command exits.

FAT 2 uses `fat identify` for this step and doesn't need `firmadyne.config`. The [current setup instructions](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#quick-start) cover the replacement.

Closing this since the old script is gone. If you hit a permission error in v2, please open a new issue with the command and path involved so I can look at that directly.

closed 2026-09-25 by adi0x90