attify/firmware-analysis-toolkit #35: qemu-system-mips: Invalid parameter 'vlan'

XT19958, 2019-10-31:
Hi
When I Setting up the network connection, there was an error.
`[?] Enter the name or absolute path of the firmware you want to analyse : /opt/firmware/Netgear.zip`
`[?] Enter the brand of the firmware : Netgear`
`[+] Now going to extract the firmware. Hold on..`
`[+] Firmware : /opt/firmware/Netgear.zip`
`[+] Brand : Netgear`
`[+] Database image ID : 1`
`[+] Identifying architecture`
`[+] Architecture : mipseb`
`[+] Storing filesystem in database`
`[!] Filesystem already exists`
`[+] Building QEMU disk image`
`[+] Setting up the network connection, please standby`
`Traceback (most recent call last):`
  ` File "./fat.py", line 122, in <module>`
    `main()`
  `File "./fat.py", line 117, in main`
    `setup_network(arch, image_id)`        
  `File "./fat.py", line 90, in setup_network`
    `child.expect("Interfaces:", timeout=None)`
  `File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect`
    `timeout, searchwindowsize, async_)`
  `File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list`
    `return exp.expect_loop(timeout)`
  `File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 117, in expect_loop`
    `return self.eof(e)`
  `File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 63, in eof`
    `raise EOF(msg)`
`pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.`
`<pexpect.pty_spawn.spawn object at 0xb7213fac>`
`command: /usr/bin/sudo`
`args: ['/usr/bin/sudo', '/opt/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh', '1', 'mipseb']`
`buffer (last 100 chars): ''`
`before (last 100 chars): "Running firmware 1: terminating after 60 secs...\r\nqemu-system-mips: Invalid parameter 'vlan'\r\nroot\r\n"`
`after: <class 'pexpect.exceptions.EOF'>`
`match: None`
`match_index: None`
`exitstatus: 1`
`flag_eof: True`
`pid: 2431`
`child_fd: 5`
`closed: False`
`timeout: 30`
`delimiter: <class 'pexpect.exceptions.EOF'>`
`logfile: None`
`logfile_read: None`
`logfile_send: None`
`maxread: 2000`
`ignorecase: False`
`searchwindowsize: None`
`delaybeforesend: 0.05`
`delayafterclose: 0.1`
`delayafterterminate: 0.1`
`searcher: searcher_re:`
    `0: re.compile('Interfaces:')`

How can I solve it?
Thank you~

extremecoders-re, 2019-10-31:
Which version of qemu are you using?

The recent versions of qemu have removed the `vlan` parameter. You can try downgrading to an earlier version (<3.0).

XT19958, 2019-11-01:
@extremecoders-re 
Thank you for you answer.I did use the latest version of qemu before.

closed 2020-03-31 by extremecoders-re