attify/firmware-analysis-toolkit #36: no IPv6 routers present

syhxsqqZM, 2019-11-08:
/ _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : wnap320.zip
[?] Enter the brand of the firmware  : wnap320.zip
[+] Now going to extract the firmware. Hold on..
[+] Firmware : wnap320.zip
[+] Brand : netgear
/home/john/firmadyne/sources/extractor/extractor.py -b netgear -sql 127.0.0.1 -np -nk "wnap320.zip" images 
[+] Database image ID : 1
[+] Identifying architecture
/home/john/firmadyne/scripts/getArch.sh ./images/1.tar.gz
[+] Architecture : mipseb
[+] Storing filesystem in database
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : [('brtrunk', '192.168.0.100')]
[+] Running the firmware finally
[+] command line : sudo /home/john/firmadyne/scratch/1/run.sh
[*] Press ENTER to run the firmware...
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 0
Bringing up TAP device...
123456
Adding route to 192.168.0.100...
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:39:21 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0061e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8f1000 @ 0070e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57792 bytes for tracking 1806 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 806aa3c0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252428k/254916k available (4260k kernel code, 2252k reserved, 1549k data, 220k init, 0k highmem)
[... 398 lines cut ...]

Starting Translator...      [dhcp]

Starting Translator...      kill: cannot kill pid 594: No such process
[ntp]

Starting Translator...      [timezone]

tarting Translator...      [sc_radio]
kill: cannot kill pid 607: No such process
Error in opening the device.
: No such device

System initilization is ..  [DONE...]
[   20.432000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 24646
[   20.436000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 24646
[   20.444000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 24646

Welcome to SDK.

Have a lot of fun...

netgear123456 login: [   23.576000] brtrunk: port 1(eth0) entering forwarding state
[   29.196000] brtrunk: no IPv6 routers present
[   29.232000] eth0: no IPv6 routers present

adi0x90, 2026-09-25:
I wouldn't dismiss this as an IPv6 message. The full log also has ext2 deleted-inode errors, missing-device failures, and a segmentation fault. Those need a closer look.

FAT 2 changes the native runtime preparation and host connection setup, but I don't have a matching result showing that those failures are gone for this WNAP320 image.

Please share its exact version/hash, the v2 backend and kernel profile, and a new serial log. I'd start with the earliest filesystem or process failure rather than the last IPv6 line.

adi0x90, 2026-09-26:
I tested a public WNAP320 `2.1.5` image after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108). Both the vendor ZIP and its TAR member now recover the rootfs, with 1,135 regular files and symlink targets matching independent reference extraction.

Your filename is `wnap320.zip`, so I still need the version or hash to confirm that it is the same release. WNAP320 also contains filenames that differ only by case; the extraction project needs case-sensitive storage, such as a Linux ext4 filesystem.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together.

I'm keeping this open for the filesystem errors and crashes in the boot log. Please include the exact firmware version and a fresh serial log when you retry; the `no IPv6 routers present` line alone does not tell us whether the firmware service is working.