attify/firmware-analysis-toolkit #37: how to use gdb in FAT

sssyyynnngithub, 2019-11-23:
Sir, May I ask you that how can I use gdb in FAT? I don not know how to use gdbserver and gdb-multiarch to debug in FAT. There are something wrong with the debug communication that I can not debug the firmware program like HTTPD. Could you please share your way to figure it out ? Thank you very much !

extremecoders-re, 2019-11-24:
You need to get hold of a statically compiled gdbserver and somehow push it to the firmware image. Like you can do this by repacking the firmware or transfer it over scp if a ssh daemon is running. After you run the program in gdbserver you should be able to connect from your host using gdb-multiarch. If this doesn't work ensure that the network is properly setup.

Another way you can debug is to use qemu user mode emulation and use the qemu gdb stub. This is a much better way than the earlier.

adi0x90, 2026-09-25:
FAT 2 now has commands for this. For a running session, I'd start by checking which debugging connections are available, then connect through GDB:

`fat debug surfaces --project ./project`

`fat debug gdb --project ./project`

Replace `./project` with your FAT project directory. You'll need GDB on the host and a compatible running session. There are also `shell`, `maps`, and `monitor` subcommands where the backend supports them; `fat debug --help` has the options.

For your HTTPD example, there's one distinction to keep in mind. The native system-mode QEMU GDB endpoint debugs the VM's CPU state. Attaching to an individual guest process needs a process-debugging path supported by the backend.

Closing this with the current commands above. If the connection fails, please open an issue with the backend, `fat debug surfaces` output, and exact GDB error so I can look at it.

closed 2026-09-25 by adi0x90