attify/firmware-analysis-toolkit #38: Duplicate ID 'net0' for netdev

zu1k, 2019-12-27:
when I run run.sh, it shows that """qemu-system-mipsel: -netdev tap,id=net0,ifname=tap1_0,script=no: Duplicate ID 'net0' for netdev"""

Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.1.1...
Starting firmware emulation... use Ctrl-a + x to exit
binenv
qemu-system-mipsel: -netdev tap,id=net0,ifname=tap1_0,script=no: Duplicate ID 'net0' for netdev
Deleting route...
Bringing down TAP device...
Deleting TAP device tap1_0...
Set 'tap1_0' nonpersistent
Set 'tap1_0' nonpersistent

extremecoders-re, 2020-02-23:
This is a bug in firmadyne.
You need to edit `run.sh` and make sure each `netdev` and `device` pair use different id's.

## Example

(Formatted for readability)

```
${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \
    -drive if=ide,format=raw,file=${IMAGE} \
    -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
    -nographic \
    -netdev socket,id=net0,listen=:2000 -device e1000,netdev=net0 \
    -netdev tap,id=net0,ifname=${TAPDEV_0},script=no -device e1000,netdev=net0 \
    -netdev socket,id=net2,listen=:2002 -device e1000,netdev=net2 \
    -netdev socket,id=net3,listen=:2003 -device e1000,netdev=net3 \
    | tee ${WORK_DIR}/qemu.final.serial.log
```
Above both first and second `netdev` and `device` pairs use the same `net0` id. To fix change the second to `net1` as shown below.

```
${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \
    -drive if=ide,format=raw,file=${IMAGE} \
    -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
    -nographic \
    -netdev socket,id=net0,listen=:2000 -device e1000,netdev=net0 \
    -netdev tap,id=net1,ifname=${TAPDEV_0},script=no -device e1000,netdev=net1 \
    -netdev socket,id=net2,listen=:2002 -device e1000,netdev=net2 \
    -netdev socket,id=net3,listen=:2003 -device e1000,netdev=net3 \
    | tee ${WORK_DIR}/qemu.final.serial.log
```

closed 2020-02-23 by extremecoders-re

alex4ord, 2020-02-23:
Thanks very much

zu1k, 2020-02-23:
Thank you very much for your reply