attify/firmware-analysis-toolkit #41: expecting for password

paramikoooo, 2020-03-12:
In this line 
https://github.com/attify/firmware-analysis-toolkit/blob/master/fat.py#L82
I got error because of pexpect wait for the password.

buffer (last 100 chars): b'Password for user firmadyne: '
before (last 100 chars): b'Password for user firmadyne: '
after: <class 'pexpect.exception.TIMEOUT'>

extremecoders-re, 2020-03-13:
Can you check if this line in scripts/getArch.sh is commented.

https://github.com/firmadyne/firmadyne/blob/52fd43d81b4cbd377fc9b2ecbdc4472027701c48/scripts/getArch.sh#L88
 
If you have installed fat using the setup.sh script, it should have commented that line.

extremecoders-re, 2020-03-13:
While installing fat using the setup.sh script, it automatically patches firmadyne such that the PostgreSQL database is not needed anymore.

closed 2020-03-31 by extremecoders-re