attify/firmware-analysis-toolkit #43: Attempts to use loopback network address (127.0.0.0) instead of a valid interface?

djmccarthy12, 2020-03-31:
Not sure what would be causing this, everything appears to be working correctly aside from the fact that it tries to spin up the connection on 127.0.0.0 (see the output below) rather than on one of the two other available interfaces.

[+] Firmware: R6950.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('lo', '127.0.0.0')]
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

extremecoders-re, 2020-04-02:
Maybe a bug in Firmadyne particularly in the script `scripts/makeNetwork.py`

Firmadyne identifies the network interface by parsing the kernel bootlog which you can find in the file `qemu.initial.serial.log`.

In the file you will find lines of the following type.

```
[    2.508000] firmadyne: __inet_insert_ifa[PID: 49 (ifconfig)]: device:lo ifa:0x0100007f
--- [ snip] ---
[    4.300000] firmadyne: __inet_insert_ifa[PID: 193 (ifconfig)]: device:br0 ifa:0x0101a8c0
```

In this example ,
`0x0100007f` is 127.0.0.1 little endian byte order and interface name is `lo`. Firmadyne doesn't create the interface if it's address is 127.0.0.1 or 0.0.0.0.

`0x0101a8c0` is 192.168.1.1 and the interface name is `br0`. Firmadyne will create this interface.

Likewise, you can go through the file and find all those lines containing the word `__inet_insert_ifa`.

extremecoders-re, 2020-04-02:
Corresponding code in [makeNetwork.py](https://github.com/firmadyne/firmadyne/blob/52fd43d81b4cbd377fc9b2ecbdc4472027701c48/scripts/makeNetwork.py#L84)

```
# Get the netwokr interfaces in the router, except 127.0.0.1
def findNonLoInterfaces(data, endianness):
    #lines = data.split("\r\n")
    lines = stripTimestamps(data)
    candidates = filter(lambda l: l.startswith("__inet_insert_ifa"), lines) # logs for the inconfig process
    if debug:
        print("Candidate ifaces: %r" % candidates)
    result = []
    if endianness == "eb":
        fmt = ">I"
    elif endianness == "el":
        fmt = "<I"
    for c in candidates:
        g = re.match(r"^__inet_insert_ifa\[[^\]]+\]: device:([^ ]+) ifa:0x([0-9a-f]+)", c)
        if g:
            (iface, addr) = g.groups()
            addr = socket.inet_ntoa(struct.pack(fmt, int(addr, 16)))
            if addr != "127.0.0.1" and addr != "0.0.0.0":
                result.append((iface, addr))
    return result
```

djmccarthy12, 2020-05-21:
Hey there - Apologies for the delay in getting back here - life got in the way. I reinstalled FAT and tried this again, having the same issue. I went through the log and found only two lines that matched the above:

[    6.160000] firmadyne: __inet_insert_ifa[PID: 198 (ifconfig)]: device:lo ifa:0x0000007f
[    6.160000] firmadyne: __inet_insert_ifa[PID: 198 (ifconfig)]: device:lo ifa:0x0100007f

Seems like its unable to find an interfact other than lo. Here is my ifconfig output:

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp0s3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:5f:52:61 brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.15/24 brd 10.0.2.255 scope global dynamic noprefixroute enp0s3
       valid_lft 79852sec preferred_lft 79852sec
    inet6 fe80::f2d7:f469:27ec:4c2e/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: enp0s8: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:82:26:90 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.4/24 brd 192.168.1.255 scope global dynamic noprefixroute enp0s8
       valid_lft 79853sec preferred_lft 79853sec
    inet6 fe80::7fe8:32c1:4f6f:372e/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

k4m4, 2022-11-12:
I was able to get it working by updating my `/etc/hosts` config such that localhost was mapped to some IP address other than `127.0.0.1`.

Abib-web, 2023-01-08:
@k4m4 I would like to know how you updated /etc/hosts? I always get br0 0x000000

adi0x90, 2026-09-25:
An inferred address of `('lo', '127.0.0.0')` wasn't a useful way to reach the firmware. I can see why the output was confusing.

FAT 2's native `qemu-direct` system mode uses QEMU user networking and port forwarding. You connect to a configured service through its mapped `127.0.0.1:<port>` endpoint. The guest still needs working networking and a service listening on the other end.

Closing this since that old address-inference step has been replaced. If the mapped endpoint doesn't work in v2, please include the firmware version, backend, and boot/service logs in a new issue.

closed 2026-09-25 by adi0x90