attify/firmware-analysis-toolkit #44: Image extraction failed

kismp123, 2020-04-02:
Hi there!, first I wanna thank you guys gave me to use this great thing.
I have met a problem as title, it messaged out 'Image extraction failed' when I executed fat.py
```
/firmware/firmware-analysis-toolkit$ ./fat.py ../WF2785_v1.8.36.bin 
                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: WF2785_v1.8.36.bin
[+] Extracting the firmware...
[!] Image extraction failed

```

I have been trying some firmwares and I got same message as I wrote.
However, when I executed /firmadyne/sources/extractor/extractor.py, it look like this
```
/firmware/WF2785_v1.8.36.bin
>> MD5: db7855f9a2a0317402068839761d15f2
>> Tag: WF2785_v1.8.36.bin_db7855f9a2a0317402068839761d15f2
>> Temp: /tmp/tmp07cylyzj
>> Status: Kernel: False, Rootfs: False, Do_Kernel: True,                 Do_Rootfs: True
>> Recursing into archive ...
>>>> Squashfs filesystem, little endian, version 4.0, compression:xz, size: 4350976 bytes, 763 inodes, blocksize: 131072 bytes, created: invalid timestamp
>>>> Found Linux filesystem in /tmp/tmp07cylyzj/_WF2785_v1.8.36.bin.extracted/squashfs-root!
>>>> gzip compressed data, maximum compression, from Unix, last modified: 2020-02-26 03:44:26
>>>> Found Linux filesystem in /tmp/tmp07cylyzj/_WF2785_v1.8.36.bin.extracted/squashfs-root!
>> Cleaning up /tmp/tmp07cylyzj...
```

I think it works.

extremecoders-re, 2020-04-02:
Please run `reset.py` or manually delete the contents of the `scratch` and `images` directory.

Tested and it works fine.
```
attify@attify:~/firmware-analysis-toolkit$ ./fat.py WF2785_v1.8.36.bin

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: WF2785_v1.8.36.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.1.1')]
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
```

kismp123, 2020-04-02:
It does not work :( 
```
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ls
LICENSE    WF2785_v1.8.36.bin  fat.config  firmadyne    reset.py
README.md  binwalk             fat.py      qemu-builds  setup.sh
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ls ..
20181005_Zyxel_LTE3302-M432_V1.00(ABKQ.1)C0.bin  firmware-analysis-toolkit
XNB-8000_1.40.02_20191024_R484.img
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ls ../../
firmware
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ls
LICENSE    WF2785_v1.8.36.bin  fat.config  firmadyne    reset.py
README.md  binwalk             fat.py      qemu-builds  setup.sh
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ./fat.py WF2785_v1.8.36.bin 
                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: WF2785_v1.8.36.bin
[+] Extracting the firmware...
[!] Image extraction failed
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ls
LICENSE    WF2785_v1.8.36.bin  fat.config  firmadyne    reset.py
README.md  binwalk             fat.py      qemu-builds  setup.sh
gegul@unknown:~/firmware/firmware-analysis-toolkit$
```
there is no images or scratch 

```
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ./reset.py
[+] Cleaning previous images and created files by firmadyne
[+] All done. Go ahead and run fat.py to continue firmware analysis
gegul@unknown:~/firmware/firmware-analysis-toolkit$ ./fat.py WF2785_v1.8.36.bin 
                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: WF2785_v1.8.36.bin
[+] Extracting the firmware...
[!] Image extraction failed
gegul@unknown:~/firmware/firmware-analysis-toolkit$
```
same.

extremecoders-re, 2020-04-02:
The `images` and `scratch` directory are located within `firmware-analysis-toolkit/firmadyne`

extremecoders-re, 2020-04-02:
Just delete the contents of those two directories. Not the directories themselves.

kismp123, 2020-04-02:
It works! Thanks!
```
gegul@unknown:~/firmware/firmware-analysis-toolkit$ sudo ./fat.py WF2785_v1.8.36.bin

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: WF2785_v1.8.36.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.1.1')]
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
```

closed 2020-04-02 by kismp123