attify/firmware-analysis-toolkit #45: [question] How can I run this firmwares?

kismp123, 2020-04-02:
Nowadays, we can find most of firmwares are encrypted.. :( 

```
gegul@unknown:~/firmware$ binwalk XNB-8000_1.40.02_20191024_R484.img
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             OpenSSL encryption, salted, salt: 0x5E8D51B0FE9D135F
22995601      0x15EE291       MySQL MISAM compressed data file Version 5
42268529      0x284F771       MySQL ISAM index file Version 11
```

Is there any idea to decrypt or run this firmware ?
or must run firmware with device?

extremecoders-re, 2020-04-02:
The issues are for FAT related issues only.  For general related discussion please go to Firmadyne.

closed 2020-04-02 by extremecoders-re

kismp123, 2020-04-02:
I mean FAT could not run encrypted firmware ? right?
```
gegul@unknown:~/firmware/firmware-analysis-toolkit$ sudo ./fat.py ../XNB-8000_1.40.02_20191024_R484.img

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme

[+] Firmware: XNB-8000_1.40.02_20191024_R484.img
[+] Extracting the firmware...
[!] Image extraction failed
```

extremecoders-re, 2020-04-02:
FAT cannot work with encrypted firmware since decrypting firmware is not a task of FAT nor of Firmadyne. There is no universal technique which will work for all encrypted firmware binaries.

This is a manual process and you need to research more about the device. If you are lucky somebody may have already done the research and published his/her findings.