attify/firmware-analysis-toolkit #46: Doesn't give an Network IP

kolet, 2020-05-09:
hey there,
im not getting any Network interface ip.

i tried multiple setup.

(vmware) LTS 18.04, 20
and linux sub system.
none of them are giving an ip

i tried multiple firmware i found online .
and no one of them are giving an ip.
not sure what to do.

[image]

also tried to go there https://github.com/firmadyne/firmadyne

and follow the guide , it did not work aswell.

extremecoders-re, 2020-05-14:
> Unfortunately, not all firmware images work out of the box. Manual modification may be needed on a case by case basis.
> 
> First try increasing the timeout value as mentioned in [Additional Notes](https://github.com/attify/firmware-analysis-toolkit#additional-notes).
> 
> If that doesn't work, you can check the kernel boot log to see if at all it assigns an IP to an interface.
> [attify/firmware-analysis-toolkit#43 (comment)](https://github.com/attify/firmware-analysis-toolkit/issues/43#issuecomment-607651532)

Hi, I mistakenly [commented](https://github.com/firmadyne/firmadyne/issues/142#issuecomment-626529747) on the Firmadyne repo which I actually intended to do here. Since the same issue was in both places.

As said, you need to go though the file `qemu.initial.serial.log`. You can find the file under `scratch/images/<image id>/`

Search for occurrences of the word `__inet_insert_ifa` in it.

adi0x90, 2026-09-25:
The old setup tried to find a guest interface address during its first boot pass. If it didn't find one, you were left without a usable connection address, even after trying different host environments.

In FAT 2, native `qemu-direct` system mode uses host-port forwarding. The guest still needs to boot, configure networking, and start the service.

Closing this since that IP-inference setup has been replaced. If an image still has no reachable service through the [v2 workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime), please open a new issue with the firmware version, backend, and serial log.

closed 2026-09-25 by adi0x90