attify/firmware-analysis-toolkit #47: Cannot access interface through assigned IP address

Khanejo, 2020-05-22:
I am using Ubuntu 18.04LTE, The installation and setup worked lawlessly  and a IP is generated:

[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1')]

..
..
[    2.656000] 802.1Q VLAN Support v1.8 Ben Greear <greearb@candelatech.com>
[    2.660000] All bugs added by David S. Miller <davem@redhat.com>
[    2.660000] lib80211: common routines for IEEE802.11 drivers
[    2.660000] lib80211_crypt: registered algorithm 'NULL'
[    2.664000] rtc_cmos rtc_cmos: setting system clock to 2020-05-22 13:04:24 UTC (1590152664)
[    2.668000] input: AT Raw Set 2 keyboard as /devices/platform/i8042/serio0/input/input0
[    2.872000] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input1
[    2.904000] EXT2-fs warning: mounting unchecked fs, running e2fsck is recommended
[    2.904000] VFS: Mounted root (ext2 filesystem) on device 8:1.
[    2.908000] Freeing prom memory: 956k freed
[    2.924000] Freeing unused kernel memory: 220k freed
[    2.932000] Warning: unable to open an initial console.
[    2.968000] firmadyne: sys_reboot[PID: 1 (init)]: magic1:fee1dead, magic2:28121969, cmd:0
[    2.996000] firmadyne: do_execve: /firmadyne/console
[    2.996000] OFFSETS: offset of pid: 0x100 offset of comm: 0x1f0
[    7.620000] ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.624000] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[    7.624000] ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.716000] device eth0.3 entered promiscuous mode
[    7.720000] device eth0 entered promiscuous mode
[    7.720000] br0: port 1(eth0.3) entering forwarding state
[    7.728000] device eth0.4 entered promiscuous mode
[    7.728000] br0: port 2(eth0.4) entering forwarding state
[    7.736000] device eth0.5 entered promiscuous mode
[    7.736000] br0: port 3(eth0.5) entering forwarding state
[    7.748000] device eth0.6 entered promiscuous mode
[    7.748000] br0: port 4(eth0.6) entering forwarding state

But, when i try to access the ip through curl or by browser it generates a error:
**curl: (7) Failed to connect to 192.168.0.1 port 80: Connection refused**

I tried changing port to 2000,2001 also but the problem remained same. I have crosschecked all the settings and everything seems fine.

extremecoders-re, 2020-10-06:
This is a delayed response, but anyway.

`Connection refused` typically indicates: the web server is not running, running but listening on a different port, or there's a firewall blocking the connection. The former two are more likely. If you've shell access to the emulated device, you can check.

adi0x90, 2026-09-25:
I'd check whether the web server is running and listening on port 80 here. The assigned IP shows part of setup worked, but `Connection refused` doesn't tell us that the service started.

FAT 2's native `qemu-direct` system mode uses mapped loopback ports, so the old host-route setup is gone on that path. A guest service that fails to start would still leave you unable to connect.

For a v2 run, please share the firmware version, backend, mapped endpoint, and boot/service output. I need those to tell whether forwarding is failing or the guest has no service to connect to.

adi0x90, 2026-09-26:
I still need the firmware model/version or a download link to reproduce this connection-refused report. The assigned address alone does not identify the image.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh project, and share the firmware hash, backend, and serial log if the web interface is still unavailable.

I'm keeping this open because there is no image-specific boot or web-service result yet.