attify/firmware-analysis-toolkit #49: can't emulate an "armel" firmware

jackulder, 2020-07-14:
Hey there. I'm emulating an armel-based firmware but I got following errors:
```
[+] Firmware: ALEOS_4.4.6.002_LS300_OpenSIM_UpdatePack.zip
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: 
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fat.py", line 174, in <module>
    main()
  File "./fat.py", line 169, in main
    infer_network(arch, image_id, qemu_dir)
  File "./fat.py", line 115, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f4c2ef5c2e8>
command: /home/iot/tools/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh
args: ['/home/iot/tools/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh', '1', '']
buffer (last 100 chars): b''
before (last 100 chars): b'Error: Invalid architecture!\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 63653
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: "b'Interfaces:'"
```
Does anyone know why these errors occourred?

zazi0001, 2020-07-14:
use sudo

dplastico, 2020-07-14:
I have this exact same issue trying to open a firmware it's seems the error is at extracting
`[+] Firmware: wr21-8.1.0.1.zip
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fat.py", line 174, in <module>
    main()
  File "./fat.py", line 162, in main
    image_id = run_extractor(args.firm_path)
  File "./fat.py", line 60, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7ff7c9053f60>
command: sources/extractor/extractor.py
args: ['sources/extractor/extractor.py', '-np', '-nk', '/home/iot/wr21/wr21-8.1.0.1.zip', 'images']
buffer (last 100 chars): b''
before (last 100 chars): b'/usr/bin/env: \xe2\x80\x98python33\xe2\x80\x99: No such file or directory\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 127
flag_eof: True
pid: 11677
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: "b'Tag: '"
iot@attifyos ~/t/f/firmadyne> 
`

jackulder, 2020-07-15:
Are you sure `wr21-8.1.0.1.zip` exists in `/home/iot/wr21/`?

> I have this exact same issue trying to open a firmware it's seems the error is at extracting
> 
> `[+] Firmware: wr21-8.1.0.1.zip
> [+] Extracting the firmware...
> Traceback (most recent call last):
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
> s = os.read(self.child_fd, size)
> OSError: [Errno 5] Input/output error
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
> incoming = spawn.read_nonblocking(spawn.maxread, timeout)
> File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
> return super(spawn, self).read_nonblocking(size)
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
> raise EOF('End Of File (EOF). Exception style platform.')
> pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "./fat.py", line 174, in
> main()
> File "./fat.py", line 162, in main
> image_id = run_extractor(args.firm_path)
> File "./fat.py", line 60, in run_extractor
> child.expect_exact("Tag: ")
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
> return exp.expect_loop(timeout)
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
> return self.eof(e)
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
> raise EOF(msg)
> pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
> <pexpect.pty_spawn.spawn object at 0x7ff7c9053f60>
> command: sources/extractor/extractor.py
> args: ['sources/extractor/extractor.py', '-np', '-nk', '/home/iot/wr21/wr21-8.1.0.1.zip', 'images']
> buffer (last 100 chars): b''
> before (last 100 chars): b'/usr/bin/env: \xe2\x80\x98python33\xe2\x80\x99: No such file or directory\r\n'
> after: <class 'pexpect.exceptions.EOF'>
> match: None
> match_index: None
> exitstatus: 127
> flag_eof: True
> pid: 11677
> child_fd: 5
> closed: False
> timeout: None
> delimiter: <class 'pexpect.exceptions.EOF'>
> logfile: None
> logfile_read: None
> logfile_send: None
> maxread: 2000
> ignorecase: False
> searchwindowsize: None
> delaybeforesend: 0.05
> delayafterclose: 0.1
> delayafterterminate: 0.1
> searcher: searcher_string:
> 0: "b'Tag: '"
> iot@attifyos ~/t/f/firmadyne>
> `
Are you sure `wr21-8.1.0.1.zip` exists in `/home/iot/wr21/`?

sud0why, 2020-07-27:
The same problem..

painnagato250, 2020-08-16:
HI dear friends i have the same problem when try to launch fat.py with my firmware how may i resold this , please somone help me

csenger18, 2020-09-16:
same issue here.

painnagato250, 2020-09-16:
so how can we fix it because i realy need to emule my firmwares

adi0x90, 2026-09-25:
The log shows an empty architecture being passed to `inferNetwork.sh`, which then returns `Invalid architecture`. So I wouldn't take this particular failure as evidence that armel emulation itself is broken.

FAT 2 separates identification from emulation planning and uses architecture, kernel, and backend profiles. Start with `fat identify`, then follow the [project workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime).

Closing this since the old script has been replaced. I still need a matching result for the ALEOS/LS300 image; if it fails in v2, please include the identification output, backend/kernel profile, and first failing step in a new issue.

closed 2026-09-25 by adi0x90