attify/firmware-analysis-toolkit #50: Image extraction failed

CaledoniaProject, 2020-07-29:
I'm trying to emulate the firmware provided in [your blog post](https://blog.attify.com/getting-started-with-firmware-emulation/)

And it failed somewhere:

```
[+] Firmware: WNAP320.zip
[+] Extracting the firmware...
[!] Image extraction failed
```

What can I do now?

extremecoders-re, 2020-07-31:
Are you using the latest Firmware Analysis Toolkit? FAT has been updated since the blog post. In particular the PostGreSQL   dependency has been removed.

CaledoniaProject, 2020-07-31:
Steps to reproduce:

1. Start Ubuntu 20.04
2. Download master branch and execute setup.sh

And I see this error (Ubuntu is shipped with python3 now)

```
sudo: pip: command not found
```

And in setup.sh

```
sudo -H pip3 install git+https://github.com/ahupp/python-magic
sudo -H pip install git+https://github.com/sviehb/jefferson
cd ..

extremecoders-re, 2020-08-01:
Thank you for the detailed steps. I was able to reproduce the issue.

The main reason is because several Python 2 related packages have been removed from Ubuntu 20.04. Unfortunately we cannot just switch to Python 3 yet because some of the dependencies of firmadyne specifically require Python 2. For example, while installing Binwalk manually, it immediately complained about the following missing packages.

```
python-pip
cramfsprogs
python-lzo
python-lzma
```

So as of now you have to stick to Ubuntu 18.04 or lower.

CaledoniaProject, 2020-08-01:
This is still needed on Ubuntu 16.04

```
apt install -y unzip python-pip
pip install pexpect
```

CaledoniaProject, 2020-08-01:
Also, please consider add `set -e` to the script

extremecoders-re, 2020-08-01:
Added `set -e`

`unzip` and `python-pip` are already specified.
https://github.com/attify/firmware-analysis-toolkit/blob/946255df4e1460508794e1af772c4ab1b543f110/setup.sh#L5

We are using `python3-pexpect` installed by `apt`.

Just tested on a 16.04 LXC container and it's working as expected.

CaledoniaProject, 2020-08-01:
There is a python error during installation, but seems harmless:

```
Collecting matplotlib
  Downloading https://files.pythonhosted.org/packages/7d/d4/e4c40c62cd8608ca09f0684e64139c56512e195f2351ac41a472d4dc8b38/matplotlib-3.3.0.tar.gz (38.8MB)
    100% |████████████████████████████████| 38.8MB 47kB/s
    Complete output from command python setup.py egg_info:

    Beginning with Matplotlib 3.1, Python 3.6 or above is required.
    You are using Python 3.5.2.

    This may be due to an out of date pip.

    Make sure you have pip >= 9.0.1.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-hzk7vw11/matplotlib/
```

Everything worked today. I was on a minimal Ubuntu Server 16.04 installation.

Thanks!

closed 2020-08-01 by CaledoniaProject

jollytrivedi2019, 2023-07-10:
I am getting the same error while I am trying to extract the Firmware.