attify/firmware-analysis-toolkit #51: Installation script running on Ubuntu 20.04 has some issues with the packages

rotpjw, 2020-08-06:
If you can check that issue and try to solve it because I really like this tool.

extremecoders-re, 2020-08-07:
Yes this is a known issue. Please see https://github.com/attify/firmware-analysis-toolkit/issues/50

adi0x90, 2026-09-25:
This was a problem with the package list in the old `setup.sh`. Sorry it got in the way of getting FAT running.

FAT 2 uses `scripts/install.sh` to build the Rust CLI with Cargo and install under `~/.local` by default. It still needs native build dependencies; they're listed in the [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md). After installation, `fat doctor` checks which external tools are available.

Closing this since the old installer has been replaced. If you hit a problem with the new one, please open an issue with your OS version, command, and error output.

closed 2026-09-25 by adi0x90