attify/firmware-analysis-toolkit #54: cant access IP adress through browser

naveen66611, 2020-09-12:
I was trying to emulate DIR300.bin binary ...

Following is the console output. I cant access the IP through browser. Help will be very much appreciated.

Welcome to fish, the friendly interactive shell
iot@attifyos ~/t/firmware-analysis-toolkit> ./fat.py DIR_firmware.bin 

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: DIR_firmware.bin
[+] Extracting the firmware...
[+] Image ID: 3
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1')]
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/iot/tools/firmware-analysis-toolkit/firmadyne/scratch/3/run.sh
[sudo] password for iot: 
Creating TAP device tap3_0...
Set 'tap3_0' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.0.1...
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:39:21 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0061e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8f1000 @ 0070e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57792 bytes for tracking 1806 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 806aa3c0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[... 427 lines cut ...]
Start Neap Server ...
start igmpproxy ...
[/etc/templates/igmpproxy.sh] start ...
[/var/run/igmpproxy_stop.sh] ...
Stop IGMP proxy ...
/proc/net/br_igmpp_br0: cannot create
/proc/net/br_mac_br0[/etc/templates/igmpproxy_helper.sh] flush ...
: cannot create
[/var/run/igmpproxy_helper.sh] ...
[/etc/templates/rg.sh] misc ...
[/var/run/rg_misc.sh] ...
/proc/fastnat/forskipsupport: cannot create
[/var/run/igmpproxy_start.sh] ...
Start IGMP proxy ...
/proc/net/br_igmpp_br0: cannot create
/proc/net/br_mac_br0: cannot create
/proc/net/br_igmpp_br0: cannot create
[/etc/init.d/S10system.sh] done!
rcS done!
Starting pid 943, console /dev/tts/0: '/bin/sh'

BusyBox v1.00 (2010.11.25-09:44+0000) Built-in shell (msh)
Enter 'help' for a list of built-in commands.

# [   52.540000] eth0.2: no IPv6 routers present

extremecoders-re, 2020-10-06:
Please check `qemu.initial.serial.log`. There should be lines containing `__inet_insert_ifa`. If not present, it indicates the firmware doesn't try to bring up the network interface and consequently there's no IP assigned.

More details https://github.com/attify/firmware-analysis-toolkit/issues/43#issuecomment-607651532

adi0x90, 2026-09-25:
Your boot log includes errors around device-specific `/proc` entries. I'd want to check those before assuming the DIR-300 web interface is only blocked by the host network setup.

FAT 2's native `qemu-direct` system mode uses port forwarding, and the [current workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) separates extraction, analysis, and preflight from launch. That still leaves the question of whether this firmware starts its web service.

Please add the exact DIR-300 version, backend, serial log, and mapped service endpoint from a v2 run so I can see where startup stops.

adi0x90, 2026-09-26:
The extraction improvements in [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) now recover a public DIR-300 A1 `1.05b09` image that failed in the earlier verification pass. Its 842 regular files and symlink targets matched independent reference extraction. These extraction checks used case-sensitive APFS.

Your report uses `DIR300.bin` / `DIR_firmware.bin`, so I cannot yet match that result to your exact hardware revision and firmware version. Please share those details or a hash/download link.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'm leaving this open for the browser-access failure. Once the image is matched, a fresh serial log and the backend used will help check whether the guest web server starts and stays available.