attify/firmware-analysis-toolkit #55: Missing Jefferson dependecies after install

csenger18, 2020-09-19:
Please update setup.sh
CLI output:

~/firmware-analysis-toolkit$ jefferson 
Traceback (most recent call last):
  File "/usr/local/bin/jefferson", line 10, in <module>
    import cstruct
ImportError: No module named cstruct
#######################################
Dependencies

    cstruct
    pylzma

source:https://github.com/sviehb/jefferson

csenger18, 2020-09-19:
But after git cloned and installed with  guide on his site  is working for me.

adi0x90, 2026-09-25:
The missing `cstruct` import explains why Jefferson wouldn't run after installation. Thanks for tracking down the dependencies and sharing what worked.

FAT 2's installer no longer installs Jefferson or manages its Python dependencies. You can follow the [current extraction setup](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md#external-tools) and use `fat doctor` to check the tools it needs.

Closing this since that install step is gone. If you're still having trouble extracting a JFFS2 image in v2, please open a new issue with the extraction command and output.

closed 2026-09-25 by adi0x90