attify/firmware-analysis-toolkit #56: "python-pip candidate not available" error while installing.

shaaa9, 2020-10-02:
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate

extremecoders-re, 2020-10-06:
This is a known issue. Python2 has been removed from Ubuntu 20.04. However we cannot switch to 20.04 yet. Some dependencies rely on Python 2. 

Please check https://github.com/attify/firmware-analysis-toolkit/issues/50#issuecomment-667496219

h2th3k, 2021-02-12:
Same issue spotted. Any way around this?

adi0x90, 2026-09-25:
The old installer asked for Python 2's `python-pip` package, which distributions had removed. That's why installation stopped here.

FAT 2 no longer needs that package. The CLI uses Rust/Cargo, though Python 3 is still needed to build the runtime-data bundle. Some optional tools also have Python requirements; the [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md) lists them.

Closing this since the Python 2 install step has been removed. You don't need to bring back `python-pip` to install v2.

closed 2026-09-25 by adi0x90