attify/firmware-analysis-toolkit #59: Trying to emulate IP cameras

josephKhoury95, 2020-11-04:
Hello,

I am interested in using firmware-analysis-toolkit to emulate IP cameras, is it possible using this tool?
If yes can you please suggest some firmware to test, because I am still not able to emulate IP cameras...

Thank you

extremecoders-re, 2020-11-09:
It should be possible to firmware emulate of IP camera. However it's not always a straightforward process.

Did you try to extract the firmware using binwalk?

josephKhoury95, 2020-11-09:
Yes, I know it is not a straight forward process.

Yes, I did use Binwalk to extract the firmware binary file. But also Attify has Binwalk in its steps, right?

@extremecoders-re, since you are a contributor to this project. I also have a question for you. Is it possible to emulate multiple firmware at the same time using Attify? I know I need to edit a lot of files to make it work, but is it possible from your point of view?

Thank you

extremecoders-re, 2020-11-10:
Yes, firmadyne and fat does use binwalk. My point was to try out binwalk separately and check the files it extracts. If binwalk doesn't work proper in the first place, firmadyne won't work too.

Emulating multiple firmware's is possible. Each firmware get's a different tap interface. As long as this condition is satisfied it should work.

To test, you can also try emulating the same firmware twice at the same time.  FAT doesn't use the SQL database, so they will be assigned different ID's even if they are the same files.

extremecoders-re, 2020-11-10:
Another possible issue that just came to my mind is conflict of IP address. By default most of the firmware use the `192.168.0.0/24` address. Firmadyne set's up a route so that the traffic gets routed properly to the tap interface. The same address cannot route to multiple interfaces. This issue has to be sorted out.

adi0x90, 2026-09-25:
I'd start with one camera model and the service you want to run. Extracting its files, starting a web server, and reproducing the whole camera runtime can involve quite different work.

FAT 2 separates extraction and analysis from emulation. Its native system runner can use port forwarding, but camera services may still need device drivers, extra partitions, or vendor-specific startup behavior.

Please share the camera model, firmware version, target service, and output from the [v2 workflow](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime). That gives me a specific case to assess.

adi0x90, 2026-09-26:
I still need a camera model and firmware release to check this report. [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) includes the nested-container and older SquashFS extraction changes from [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108).

Please share one camera model, its firmware download/version, and the service you want to run, such as the web interface or video stream. I can then separate extraction support from the device-specific work needed to start that service. I'll keep this open until there is a concrete target and result.