attify/firmware-analysis-toolkit #60: Problem trying connecting to HTTP server in TP-Link TL-WR940N v6

tati1206, 2020-11-05:
When I create an instance of the firmware, I can't connect to the HTTP server. I was making some test, and when I use the ping utility to see if the firmware is receiving packets (`ping 192.168.0.1`), it doesn't send any response. Then I tried with other utility called arping to send ARP packets, because I can explicitly specify the interface where I want to send the packets (`arping -I tap1_0 192.168.0.1`), and with that the firmware was sending responses. 

I also tested with nmap to see if it detects some open ports, and I didn't get something different, all the ports were detected as closed (`nmap -Pn 192.168.0.1`). 

Ifconfig output from the router firmware.
```
# ifconfig
br0       Link encap:Ethernet  HWaddr 00:0A:EB:13:7B:00  
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:46 errors:0 dropped:0 overruns:0 frame:0
          TX packets:136 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:5034 (4.9 KiB)  TX bytes:25984 (25.3 KiB)

eth0      Link encap:Ethernet  HWaddr 00:0A:EB:13:7B:00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:48 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5798 (5.6 KiB)  TX bytes:3582 (3.4 KiB)

eth1      Link encap:Ethernet  HWaddr 52:54:00:12:34:57  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:9318 (9.0 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:20 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1880 (1.8 KiB)  TX bytes:1880 (1.8 KiB)
```

And the logs of the firmware init.

```
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.39.4+ (ddcc@ddcc-virtual) (gcc version 5.3.0 (GCC) ) #2 Tue Sep 1 18:08:53 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 00678000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f897000 @ 00768000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 60672 bytes for tracking 1896 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 80702800, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[... 289 lines cut ...]
[    3.916000] ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[    4.020000] br0: port 1(eth0) entering forwarding state
[    4.020000] br0: port 1(eth0) entering forwarding state

 (none) mips #2 Tue Sep 1 18:08:53 EDT 2020 (none)
(none) login: [   14.240000] eth0: no IPv6 routers present
[   14.320000] eth1: no IPv6 routers present
[   14.928000] br0: no IPv6 routers present

 (none) mips #2 Tue Sep 1 18:08:53 EDT 2020 (none)
(none) login: 
 (none) mips #2 Tue Sep 1 18:08:53 EDT 2020 (none)
(none) login: root 
Password: 
Login incorrect
(none) login: [   93.068000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 722
[   93.084000] kernel msg: ebtables bug: please report to author: Wrong len argument
exec_Cmd: OPEN ALL PHY ETH!!
 
exec_Cmd: [HTTPS]httpsImplHelp.c:httpsSslPhraseHandle,557 SSL_CTX_use_certificate_file success! 
exec_Cmd: [HTTPS]httpsImplHelp.c:httpsSslPhraseHandle,572 SSL_CTX_use_PrivateKey_file success! 

TL-WR940N login: root
Password:
```

extremecoders-re, 2020-11-06:
You've have to experiment and try various options.

From the log, it looks like Ethernet frames are indeed being received by the emulated firmware. The IP layer is not working due to some issue.

I would suggest to put socat or netcat within the firmware itself. You can delete the bridge `br0`. 
Set up a netcat/socat listener on `eth0` or `eth1`within the firmware and try to connect to it from the host. If no IP is assigned automatically to those interfaces you can set one manually. Alternatively, there may be `dhclient` or similar within the firmware which you can use.

tati1206, 2020-11-06:
mmm, how can I send the netcat executable to the firmware? Also, trying to share the netcat executable using a tftp client that comes with the firmware, I realized that the firmware is sending IP packets with a source address of 0.0.0.0 

Also, I tried with the dropbear ssh server that comes with the firmware to create a listener into the firmware and try to connect from my computer, and still happening the same, the firmware is receiving the packet, but its not responding, as you told, only packets of the link layer works, but then IP layer don't.

extremecoders-re, 2020-11-06:
You can use the [`mount.sh`](https://github.com/firmadyne/firmadyne/blob/master/scripts/mount.sh) that comes with firmadyne to mount the firmware filesystem on your local machine. There you should be able to transfer the netcat executable to `/bin`, unmount it and rerun the firmware.

tati1206, 2020-11-06:
Thanks fro the tips!, I didn't know that.
Okay, I created a listener with netcat, and connected from the host PC, and as I expected didn't work. It doesnt even do the three way handshake, the host only sends the SYN packet, and the firmware don't send anything.

tati1206, 2020-11-06:
mmm, all the packets on the IP layer has a source IP of 0.0.0.0
[image]

extremecoders-re, 2020-11-07:
Hmm, you try deleting the bridge `br0` and set up the listener on either `eth0` or `eth1`. If they don't have an IP manually assign them using `ifconfig`, make sure they are on the same subnet as the host tap interface.

tati1206, 2020-11-07:
Yeah, I tried all that. Something I discovered, is that putting a different IP, like 192.168.3.0/24 instead of 192.168.0.0/24, now the source IP its good, and now its not 0.0.0.0. But still, the packets are not received by the firmware, like, receive the frame, but the IP layer don't handle it.

tati1206, 2020-11-07:
omg, I get it to work. It was iptables. I only deleted all entries and put default entries to accept all inbound connections, now it works. Thanks for the help! I learned more about this.

closed 2020-11-07 by tati1206

Lexxie4, 2021-04-13:
Dear tati1454,
I'm currently having the same problem with the TP-Link c7 firmware emulation in Firmadyne. Could you please give me a bit more information about changing iptables. My iptable-xml file is located in root/bin/ and has a broken symbolic link to /lzm/qca_dual/build/ ../rootfs.build.2.6.31_ap135v2_wifi/sbin/iptables-multi. Would really appreciate it!

tati1454, 2021-04-13:
I think your problem its a bit more complicated but, if the iptables cli works, I runned `iptabled -F` to flush all iptables entries and then `iptables -P INPUT ACCEPT` to accept all input packets.

Lexxie4, 2021-04-13:
I had to mention that I only see the broken symlink after extracting the binary with binwalk -e. 
After I logged in as root in Firmadyne and used `iptables -F` and `iptables -P INPUT ACCEPT` it worked!! Thank for your quick response and help.

15890088059, 2022-09-20:
Can the simulated wr940n log in to the terminal shell, and what is the login password

tati1454, 2022-09-21:
Yes, if I remember correctly the passowrd is `shoadmin` and the user `root`

AnduinBrian, 2023-12-21:
hello can you ping from emulated (192.168.0.1) to pc client (192.168.0.2). Im able to ping from 192.168.0.2 to 192.168.0.1, but cant if from 192.168.0.1 to 192.168.0.2