attify/firmware-analysis-toolkit #62: pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

terorr, 2021-03-12:
./fat.py wnap320_V3.7.11.4_firmware.tar

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fat.py", line 170, in <module>
    main()
  File "./fat.py", line 158, in main
    image_id = run_extractor(args.firm_path)
  File "./fat.py", line 52, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 418, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f994ab87040>
command: /home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py
args: ['/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py', '-np', '-nk', 'wnap320_V3.7.11.4_firmware.tar', '/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/images']
buffer (last 100 chars): b''
before (last 100 chars): b'firmware-analysis-toolkit-master/firmadyne/wnap320_V3.7.11.4_firmware.tar\r\n>> Skipping: completed!\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 23010
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Tag: '

JK2210, 2021-03-16:
Same problem.
Firmware image: https://static.tp-link.com/TL-WR945N(EU)_V1_161028.zip

cloud0606, 2021-03-23:
same problem after  i used it many times. 
firmware:  D-Link After i used it many times
``` bash
/home/cuc/tools/firmware-analysis-toolkit/fat.py  '/home/cuc/tools/firmware-analysis-toolkit/share/test/DIR825B1_FW2.05WWB07Beta06.bin'

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: DIR825B1_FW2.05WWB07Beta06.bin
[+] Extracting the firmware...
[+] Image ID: 10
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 144, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 97, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 455, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 149, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cuc/tools/firmware-analysis-toolkit/fat.py", line 166, in <module>
    main()
  File "/home/cuc/tools/firmware-analysis-toolkit/fat.py", line 161, in main
    infer_network(arch, image_id, qemu_dir)
  File "/home/cuc/tools/firmware-analysis-toolkit/fat.py", line 107, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 384, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 102, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 49, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f49d4205470>
command: /home/cuc/tools/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh
args: ['/home/cuc/tools/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh', '10', 'mipseb']
searcher: None
buffer (last 100 chars): b''
before (last 100 chars): b"use\r\nqemu-system-mips: -netdev socket,id=s1,listen=:2001: Device 'socket' could not be initialized\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 12278
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1

```

cloud0606, 2021-03-23:
It work for me killing fat-related processes, but I don’t know what caused the error, I guess it might be the memory space too small.

``` bash
cuc@cuc-VirtualBox:~/tools/firmware-analysis-toolkit$ ps aux | grep fat 
cuc       2304 82.4  0.6  53924  6348 ?        R    17:24 147:52 python3 /home/cuc/tools/firmware-analysis-toolkit/fat.py /home/cuc/tools/firmware-analysis-toolkit/share/dlink/DCS-930L_Ax_v1.15.04.bin
root      2623  0.3  6.4 1191064 64876 pts/18  Sl+  17:26   0:42 qemu-system-mipsel -m 256 -M malta -kernel /home/cuc/tools/firmware-analysis-toolkit/firmadyne/binaries//vmlinux.mipsel -drive if=ide,format=raw,file=/home/cuc/tools/firmware-analysis-toolkit/firmadyne/scratch//7//image.raw -append root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0 -nographic -netdev socket,id=net1,listen=:2001 -device e1000,netdev=net1 -netdev socket,id=net2,listen=:2002 -device e1000,netdev=net2 -netdev socket,id=net3,listen=:2003 -device e1000,netdev=net3
cuc      12754  0.0  0.1  15756  1088 pts/23   S+   20:24   0:00 grep --color=auto fat
cuc@cuc-VirtualBox:~/tools/firmware-analysis-toolkit$ sudo kill -9 2304
cuc@cuc-VirtualBox:~/tools/firmware-analysis-toolkit$ sudo kill -9 2623
```

> ./fat.py wnap320_V3.7.11.4_firmware.tar
> 
> ```
>                            __           _
>                           / _|         | |
>                          | |_    __ _  | |_
>                          |  _|  / _` | | __|
>                          | |   | (_| | | |_
>                          |_|    \__,_|  \__|
> 
>             Welcome to the Firmware Analysis Toolkit - v0.3
> Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
>               By Attify - https://attify.com  | @attifyme
> ```
> 
> [+] Firmware: wnap320_V3.7.11.4_firmware.tar
> [+] Extracting the firmware...
> Traceback (most recent call last):
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
> s = os.read(self.child_fd, size)
> OSError: [Errno 5] Input/output error
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
> incoming = spawn.read_nonblocking(spawn.maxread, timeout)
> File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
> return super(spawn, self).read_nonblocking(size)
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
> raise EOF('End Of File (EOF). Exception style platform.')
> pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "./fat.py", line 170, in
> main()
> File "./fat.py", line 158, in main
> image_id = run_extractor(args.firm_path)
> File "./fat.py", line 52, in run_extractor
> child.expect_exact("Tag: ")
> File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 418, in expect_exact
> return exp.expect_loop(timeout)
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
> return self.eof(e)
> File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
> raise EOF(msg)
> pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
> <pexpect.pty_spawn.spawn object at 0x7f994ab87040>
> command: /home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py
> args: ['/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py', '-np', '-nk', 'wnap320_V3.7.11.4_firmware.tar', '/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/images']
> buffer (last 100 chars): b''
> before (last 100 chars): b'firmware-analysis-toolkit-master/firmadyne/wnap320_V3.7.11.4_firmware.tar\r\n>> Skipping: completed!\r\n'
> after: <class 'pexpect.exceptions.EOF'>
> match: None
> match_index: None
> exitstatus: None
> flag_eof: True
[... 67 lines cut ...]
> command: /home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py
> args: ['/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/sources/extractor/extractor.py', '-np', '-nk', 'wnap320_V3.7.11.4_firmware.tar', '/home/hjk/soft/firmware-analysis-toolkit-master/firmadyne/images']
> buffer (last 100 chars): b''
> before (last 100 chars): b'firmware-analysis-toolkit-master/firmadyne/wnap320_V3.7.11.4_firmware.tar\r\n>> Skipping: completed!\r\n'
> after: <class 'pexpect.exceptions.EOF'>
> match: None
> match_index: None
> exitstatus: None
> flag_eof: True
> pid: 23010
> child_fd: 5
> closed: False
> timeout: None
> delimiter: <class 'pexpect.exceptions.EOF'>
> logfile: None
> logfile_read: None
> logfile_send: None
> maxread: 2000
> ignorecase: False
> searchwindowsize: None
> delaybeforesend: 0.05
> delayafterclose: 0.1
> delayafterterminate: 0.1
> searcher: searcher_string:
> 0: b'Tag: '

So4ms, 2021-05-14:
I encountered the same problem, have you solved it?

Applebois, 2021-09-04:
Any solutions on this ??

Applebois, 2021-09-04:
I think I found the solution.

Go to firmadyne/sources/extractor/exrtractor.py
change first line to 
#!/usr/bin/env python3

flamecopper, 2022-02-27:
For me I just reinstall the entire thing, it works for me, I swear

AblePart, 2022-04-18:
first run `./reset.py` or `sudo ./reset.py`
then `sudo ./fat.py fimwarefile.bin`

some times previous workspace lives and causes problem!

Grolar1337, 2023-02-04:
i have the same problem is there any way to fix it

KoelhoSec, 2023-04-18:
> 

> first run `./reset.py` or `sudo ./reset.py` then `sudo ./fat.py fimwarefile.bin`
> 
> some times previous workspace lives and causes problem!

Worked for me!

sdahiasdba, 2024-03-31:
(ayman㉿kali)-[~/Downloads/firmware-analysis-toolkit/firmadyne]
└─$ sudo ./fat.py Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade\(0322113411\).en_upgrade

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade(0322113411).en_upgrade
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 172, in <module>
    main()
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 160, in main
    image_id = run_extractor(args.firm_path)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ayman/Downloads/firmware-analysis-toolkit/firmadyne/./fat.py", line 54, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 432, in expect_exact
    return exp.expect_loop(timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
           ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f0484008050>
command: /usr/bin/sudo
args: ['/usr/bin/sudo', '--', '/home/ayman/Downloads/firmware-analysis-toolkit/sources/extractor/extractor.py', '-np', '-nk', 'Gaw5.6T02-4-DL-R1B020-ME.EN_upgrade(0322113411).en_upgrade', '/home/ayman/Downloads/firmware-analysis-toolkit/images']
buffer (last 100 chars): b''
before (last 100 chars): b'ayman/Downloads/firmware-analysis-toolkit/sources/extractor/extractor.py: command not found\r\nayman\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 17610
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Tag: '

I have same this problem anyone  solve it

adi0x90, 2026-09-25:
The original trace shows `Skipping: completed!` from the extractor, but `fat.py` keeps waiting for `Tag:`. The extractor has already stopped, so the wrapper ends up reporting EOF.

FAT 2 doesn't use that prompt or the old `reset.py` workflow. You can use the [project-based extraction commands](https://github.com/attify/firmware-analysis-toolkit/blob/master/README.md#image-to-runtime) instead.

Closing this since that prompt-handling code has been replaced. Thanks to everyone who shared workarounds. There are several different images in this thread; if yours still fails in v2, please open a separate issue with its version and extraction output.

closed 2026-09-25 by adi0x90