attify/firmware-analysis-toolkit #65: No route to host

blackndoor, 2021-04-22:
Just tried to run a dlink firmware:

```
./fat.py ~/Documents/mydb/formation/leaked/IoT_training/emulate_firmware/Dlink_firmware.bin                                                                                         

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: Dlink_firmware.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1')]
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/bkndr/tools/firmware-analysis-toolkit/firmadyne/scratch/1/run.sh
[sudo] Mot de passe de bkndr : 
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.0.1...
Starting firmware emulation... use Ctrl-a + x to exit
[...]
# ip a
1: lo: <LOOPBACK,UP,10000> mtu 16436 qdisc noqueue 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 52:54:00:12:34:56 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 52:54:00:12:34:57 brd ff:ff:ff:ff:ff:ff
4: eth2: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc pfifo_fast qlen 100
    link/ether 00:de:fa:19:c0:02 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2de:faff:fe19:c002/64 scope link 
       valid_lft forever preferred_lft forever
5: eth3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 52:54:00:12:34:59 brd ff:ff:ff:ff:ff:ff
6: tunl0: <NOARP> mtu 1480 qdisc noop 
    link/ipip 0.0.0.0 brd 0.0.0.0
7: sit0: <NOARP> mtu 1480 qdisc noop 
    link/sit 0.0.0.0 brd 0.0.0.0
8: ip6tnl0: <NOARP> mtu 1452 qdisc noop 
    link/tunnel6 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 brd 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
9: br0: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc noqueue 
    link/ether 00:de:fa:19:c0:01 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global br0
    inet6 fe80::2de:faff:fe19:c001/64 scope link 
       valid_lft forever preferred_lft forever
10: eth2.0@eth2: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc noqueue master br0 
    link/ether 00:de:fa:19:c0:01 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2de:faff:fe19:c001/64 scope link 
       valid_lft forever preferred_lft forever
11: eth2.2@eth2: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc noqueue 
    link/ether 00:de:fa:19:c0:02 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2de:faff:fe19:c002/64 scope link 
       valid_lft forever preferred_lft forever
# ip r
192.168.0.0/24 dev br0  proto kernel  scope link  src 192.168.0.1 
239.0.0.0/8 dev br0  scope link 
```

On my host:
```
$ ip r                                                        
default via 192.168.1.1 dev wlan0 
169.254.0.0/16 dev docker0 scope link metric 1000 linkdown 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
172.18.0.0/16 dev br-93255fe0b579 proto kernel scope link src 172.18.0.1 linkdown 
192.168.0.0/24 dev tap1_0 proto kernel scope link src 192.168.0.2 
192.168.0.1 via 192.168.0.1 dev tap1_0 
192.168.1.0/24 dev wlan0 proto kernel scope link src 192.168.1.38 
192.168.122.0/24 dev virbr0 proto kernel scope link src 192.168.122.1 linkdown 
$ curl 192.168.0.1
curl: (7) Failed to connect to 192.168.0.1 port 80: No route to host
```

tjhasan, 2021-05-25:
I am having the same issue.

adi0x90, 2026-09-25:
The original run used a Firmadyne TAP device and a host route to `192.168.0.1`. FAT 2's native `qemu-direct` system mode uses mapped host ports, but I wouldn't call this fixed yet.

The v2 test log still shows an extraction failure for a `Dlink_firmware.bin` sample with Broadcom SquashFS v3. The filename matches this report; I can't tell whether the files are identical.

Please share the model/version and image hash so I can match them. If it's the same image, extraction needs fixing before there's a filesystem to boot and a service to connect to.

adi0x90, 2026-09-26:
I now have a successful extraction result for a local `Dlink_firmware.bin` sample after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108). All 990 regular files and symlink targets matched independent reference extraction. These extraction checks used case-sensitive APFS.

The tested image has SHA-256 `632f8bdd3fc0e8415d490f7581ec13f16aa293ae412e793648fd0d3351b70fe0`. Please compare that with your file so I can confirm whether the result applies to your image.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'm keeping this open for the network/browser-access problem. If it remains after a fresh extraction and emulation attempt, please include `fat --version` output, backend, and serial log. The current result verifies extraction; guest networking still needs a runtime check.