attify/firmware-analysis-toolkit #66: Who can help me?

So4ms, 2021-05-14:
./fat.py DIR823GA1_FW102B03.bin
```bash

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: DIR823GA1_FW102B03.bin
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 144, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 97, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 455, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 149, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fat.py", line 170, in <module>
    main()
  File "./fat.py", line 158, in main
    image_id = run_extractor(args.firm_path)
  File "./fat.py", line 52, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 384, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 102, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 49, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f5d8665d2e8>
command: /home/giantbranch/Desktop/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py
args: ['/home/giantbranch/Desktop/firmware-analysis-toolkit/firmadyne/sources/extractor/extractor.py', '-np', '-nk', 'DIR823GA1_FW102B03.bin', '/home/giantbranch/Desktop/firmware-analysis-toolkit/firmadyne/images']
searcher: None
buffer (last 100 chars): b''
before (last 100 chars): b'ractor/extractor.py", line 17, in <module>\r\n    import magic\r\nImportError: No module named \'magic\'\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 3386
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1

```

1005789164, 2021-07-25:
请问楼主解决了吗？

So4ms, 2021-07-25:
> 
> 
> 请问楼主解决了吗？
没有，我是下的现成的别人弄好的虚拟机。

1005789164, 2021-07-25:
> > 好像楼主解决了？
> > 没有，我是下的现成的别人弄好的虚拟机。

所以别人的可以用了？那请问你虚拟是什么型号版本？能否给下这个虚拟机的下载地址？感谢。。

1005789164, 2021-07-25:
老哥还在吗

So4ms, 2021-07-26:
> 
> 
> 老哥还在吗

https://github.com/adi0x90/attifyos

1005789164, 2021-07-26:
> > 老哥还在吗
> 
> https://github.com/adi0x90/attifyos

谢谢，今天凌晨已经在16.04装好了。。。。。

XDTG, 2021-09-24:
> > > 老哥还在吗
> > 
> > 
> > https://github.com/adi0x90/attifyos
> 
> 谢谢，今天凌晨已经在16.04装好了。。。。。

想问下你是怎么解决的呀？

1005789164, 2021-09-25:
> > > > 老哥还在吗
> > > 
> > > 
> > > https://github.com/adi0x90/attifyos
> > 
> > 
> > 谢谢，今天凌晨已经在16.04装好了。。。。。
> 
> 想问下你是怎么解决的呀？

你可以试试这个[https://github.com/liyansong2018/firmware-analysis-plus.git](https://github.com/liyansong2018/firmware-analysis-plus.git)

adi0x90, 2026-09-25:
The underlying error is in the last bit of extractor output: `ImportError: No module named 'magic'`. The Python extractor couldn't start, and `pexpect` reported EOF afterward.

FAT 2 no longer uses that Python extractor. Follow the [current installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md), then run `fat doctor` before trying `fat extract`.

Closing this since that dependency problem belongs to the old setup. I still need a matching extraction result before calling DIR-823G supported; if it fails in v2, please share the firmware version and extraction output in a new issue.

closed 2026-09-25 by adi0x90