attify/firmware-analysis-toolkit #67: E: Unable to locate package qt5base-dev when trying to run setup.sh on a fresh install of kali linux

noobexploiter, 2021-05-15:
Hi. I have a fresh install of kali linux. While trying to run setup.sh it is giving me 
`E: Unable to locate package qt5base-dev
`

Heres the whole output of the command

```┌──(brandon㉿kali)-[~/tools/firmware-analysis-toolkit]
└─$ ./setup.sh                                                     
Hit:1 http://dl.google.com/linux/chrome/deb stable InRelease                   
Get:2 http://ftp.belnet.be/pub/kali/kali kali-rolling InRelease [30.5 kB]
Fetched 30.5 kB in 7s (4,540 B/s)                                                                                                                                                                                                           
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
567 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
busybox-static is already the newest version (1:1.30.1-6+b1).
fakeroot is already the newest version (1.25.3-1.1).
kpartx is already the newest version (0.8.5-2).
python3-pexpect is already the newest version (4.8.0-2).
python3-pip is already the newest version (20.3.4-1).
qemu-system-arm is already the newest version (1:5.2+dfsg-10).
qemu-system-mips is already the newest version (1:5.2+dfsg-10).
qemu-system-x86 is already the newest version (1:5.2+dfsg-10).
qemu-utils is already the newest version (1:5.2+dfsg-10).
snmp is already the newest version (5.9+dfsg-3+b1).
uml-utilities is already the newest version (20070815.4-1).
unzip is already the newest version (6.0-26).
util-linux is already the newest version (2.36.1-7).
vlan is already the newest version (2.0.5).
0 upgraded, 0 newly installed, 0 to remove and 567 not upgraded.
Installing binwalk
Cloning into 'binwalk'...
remote: Enumerating objects: 114, done.
remote: Counting objects: 100% (114/114), done.
remote: Compressing objects: 100% (108/108), done.
remote: Total 114 (delta 8), reused 33 (delta 1), pack-reused 0
Receiving objects: 100% (114/114), 32.87 MiB | 569.00 KiB/s, done.
Resolving deltas: 100% (8/8), done.
+ which lsb_release
+ '[' 1 -eq 0 ']'
++ lsb_release -i
++ cut -f 2
+ distro=Kali
++ lsb_release -r
++ cut -f 2
++ cut -c1-2
+ distro_version=20
+ REQUIRED_UTILS='wget tar python'
+ APTCMD=apt
+ APTGETCMD=apt-get
+ YUMCMD=yum
+ '[' Kali = Kali ']'
+ APT_CANDIDATES='git locales build-essential qt5base-dev mtd-utils gzip bzip2 tar arj lhasa p7zip p7zip-full cabextract util-linux firmware-mod-kit cramfsswap squashfs-tools zlib1g-dev liblzma-dev liblzo2-dev sleuthkit default-jdk lzop cpio'
+ PYTHON3_APT_CANDIDATES=
+ PYTHON3_YUM_CANDIDATES=
+ YUM_CANDIDATES='git gcc gcc-c++ make openssl-devel qtwebkit-devel qt-devel gzip bzip2 tar arj p7zip p7zip-plugins cabextract squashfs-tools zlib zlib-devel lzo lzo-devel xz xz-compat-libs xz-libs xz-devel xz-lzma-compat python-backports-lzma lzip pyliblzma perl-Compress-Raw-Lzma lzop srecord'
++ which python3
+ PYTHON=/usr/bin/python3
+ '[' 0 -eq 0 ']'
+ echo 'UID is 0, sudo not required'
UID is 0, sudo not required
+ SUDO=
+ '[' 1 -eq 0 ']'
+ '[' Kali '!=' Unknown ']'
+ echo 'Kali 20 detected'
Kali 20 detected
+ NEEDED_UTILS=
+ for UTIL in $REQUIRED_UTILS
+ find_path wget
+ FILE_NAME=wget
+ echo -ne 'checking for wget...'
checking for wget...+ which wget
+ '[' 0 -eq 0 ']'
+ echo yes
yes
+ return 0
+ '[' 0 -eq 1 ']'
+ for UTIL in $REQUIRED_UTILS
+ find_path tar
+ FILE_NAME=tar
+ echo -ne 'checking for tar...'
checking for tar...+ which tar
+ '[' 0 -eq 0 ']'
+ echo yes
yes
+ return 0
+ '[' 0 -eq 1 ']'
+ for UTIL in $REQUIRED_UTILS
+ find_path python
+ FILE_NAME=python
+ echo -ne 'checking for python...'
checking for python...+ which python
+ '[' 0 -eq 0 ']'
+ echo yes
yes
+ return 0
+ '[' 0 -eq 1 ']'
+ find_path apt
+ FILE_NAME=apt
+ echo -ne 'checking for apt...'
checking for apt...+ which apt
+ '[' 0 -eq 0 ']'
+ echo yes
yes
+ return 0
+ '[' 0 -eq 1 ']'
+ apt install -s -y dpkg

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

+ PKGCMD=apt
+ PKGCMD_OPTS='install -y'
+ PKG_CANDIDATES='git locales build-essential qt5base-dev mtd-utils gzip bzip2 tar arj lhasa p7zip p7zip-full cabextract util-linux firmware-mod-kit cramfsswap squashfs-tools zlib1g-dev liblzma-dev liblzo2-dev sleuthkit default-jdk lzop cpio'
+ PKG_PYTHON3_CANDIDATES=
+ '[' '' '!=' '' ']'
+ cd /tmp
+ apt install -y git locales build-essential qt5base-dev mtd-utils gzip bzip2 tar arj lhasa p7zip p7zip-full cabextract util-linux firmware-mod-kit cramfsswap squashfs-tools zlib1g-dev liblzma-dev liblzo2-dev sleuthkit default-jdk lzop cpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package qt5base-dev
```

noobexploiter, 2021-05-16:
https://www.gitmemory.com/issue/ReFirmLabs/binwalk/537/824300091 this works for me

cosad3s, 2022-10-13:
`sudo apt install qtbase5-dev` (https://github.com/ReFirmLabs/binwalk/issues/537)

master1018, 2023-04-07:
I meet the same issue when install the binwalk, maybe your kali have installed the binwalk, and you can try to modify the setup.sh, skip the install of binwalk

adi0x90, 2026-09-25:
Your log points to the old Binwalk dependency script, which asked the package manager for `qt5base-dev`. Thanks to everyone who shared workarounds here.

FAT 2's installer no longer runs that script or requests that package. The [installation guide](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md) has the current build and extraction requirements.

Closing this since the install step that caused it has been replaced. If the new installer fails on Kali, please include the Kali version and installer output in a new issue.

closed 2026-09-25 by adi0x90