attify/firmware-analysis-toolkit #68: Image Extraction Failed. Trying to Emulate Sample Arduino bin File

tjhasan, 2021-05-25:
I am attempting to emulate a sample Arduino firmware using FAT. The bin file that I am working with can be found [here](https://drive.google.com/file/d/1xcsDFyV-FC1btMUyWNDa9aZZT6ndNE_k/view?usp=sharing).

It is a very simple firmware that just blinks the light on the board. It does not create a network connection so I don't expect a network connection to actually start. If the image extraction is successful and I can run the firmware by using `./fat.py` then I will assume that the firmware has been emulated.

However, when I run the program, I get the following output:
```bash
ubuntu@ubuntu1804:~/Desktop/firmware-analysis-toolkit$ ./reset.py 
[+] Cleaning previous images and created files by firmadyne
[+] All done. Go ahead and run fat.py to continue firmware analysis
ubuntu@ubuntu1804:~/Desktop/firmware-analysis-toolkit$ sudo ./fat.py ./Images/Arduino/Example_Code.ino.nano_33_iot.bin --qemu 2.5.0
                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: Example_Code.ino.nano_33_iot.bin
[+] Extracting the firmware...
[!] Image extraction failed
ubuntu@ubuntu1804:~/Desktop/firmware-analysis-toolkit$ 
```
Is there a place where I can examine exactly why the extraction failed, or is emulating this type of firmware impossible on the FAT / Firmadyne tool? If there is a setup process for this type of firmware, where can I find the instructions for it? 

Any assistance would be greatly appreciated.

extremecoders-re, 2021-05-26:
It isn't possible to emulate arbitrary firmware with FAT.  FAT only supports firmware for ARM and MIPS devices running a Linux kernel. Arduino doesn't run Linux and is rather powered by a AVR microcontroller (which is neither ARM or MIPS).

The firmware for an Arduino is a binary blob without any filesystem. During the image extraction stage, FAT uses binwalk to try extract the firmware which fails in this case. To cross check you can try running binwalk directly on the firmware binary.

There are some online Arduino simulators though which may better fit your use case.

tjhasan, 2021-05-26:
Thank you very much for the thorough explanation. I can confirm that running binwalk directly on the binary results in no return, so that validates your statement. I had a feeling this was the case but wanted to make sure regardless.

That being said, I have seen the various Arduino simulators online, but I'm looking for an emulator as opposed to a simulator for the project that I have in mind. The problem is that there are not any well maintained AVR emulators currently online.

extremecoders-re, 2021-05-27:
Qemu 6.0 supports emulating AVR.

- https://qemu.readthedocs.io/en/latest/system/target-avr.html
- https://yeah.nah.nz/embedded/qemu-avr/

Beside also check these
- https://github.com/buserror/simavr
- https://github.com/lcgamboa/picsimlab

closed 2021-06-01 by extremecoders-re