attify/firmware-analysis-toolkit #70: There is an error in the network interface assignment.

dhje0ng, 2022-05-09:
[qemu.initial.serial.log](https://github.com/attify/firmware-analysis-toolkit/files/8649289/qemu.initial.serial.log)
[qemu.final.serial.log](https://github.com/attify/firmware-analysis-toolkit/files/8649293/qemu.final.serial.log)

If you extract the firmware through the ./fat.py script and try to emulate it, the IP is assigned to the br-lan interface of 192.168.1.254.
However, this is an IP that is no longer available after some time has elapsed during the boot process. Host is down problem.

So I checked the qemu boot log of the firmadyne tool to confirm that it was a boot process problem. Does it have something to do with the Segmentation Fault error that occurs during boot?

[image]

We need a way to solve this problem. thank you!

extremecoders-re, 2022-05-09:
> Does it have something to do with the Segmentation Fault error that occurs during boot?

Quite likely. Looks like after the device has finished booting some service is started which segfaults. The proper way to diagnose the issue is to investigate which services are launched after boot and try by disabling them one by one.

dhje0ng, 2022-05-09:
```
[    5.656000] mount_root: mounting /dev/root
[    5.660000] mount_root: loading kmods from internal overlay
[    5.672000] mount_root: failed to launch kmodloader from internal overlay
[    5.712000] do_page_fault() #2: sending SIGSEGV to block for invalid read access from
[    5.712000] 00000010 (epc == 0040638c, ra == 00403394)
[    5.712000] Cpu 0
[    5.712000] $ 0   : 00000000 1000a400 00000000 00000000
[    5.712000] $ 4   : 00000000 00000000 7f8f1214 7f8f1218
[    5.712000] $ 8   : 00000000 00000fa5 00000001 fffffff8
[    5.712000] $12   : 00000001 ffff0000 00000010 00420000
[    5.712000] $16   : 00420000 00000000 00420000 00407be4
[    5.712000] $20   : 7f8f1214 7f8f1218 0000003a 2b2c9d8c
[    5.712000] $24   : 00418ffc 2b234160
[    5.712000] $28   : 2b2cf2c0 7f8f0fc8 00000000 00403394
[    5.712000] Hi    : 0000002f
[    5.712000] Lo    : 02762762
[    5.712000] epc   : 0040638c 0x40638c
[    5.712000]     Not tainted
[    5.712000] ra    : 00403394 0x403394
[    5.712000] Status: 0000a413    USER EXL IE
[    5.712000] Cause : 10800008
[    5.712000] BadVA : 00000010
[    5.712000] PrId  : 00019300 (MIPS 24Kc)
[    5.712000] Modules linked in:
[    5.712000] Process block (pid: 134, threadinfo=8f45c000, task=8f43e4d0, tls=2b2d0d48)
[    5.712000] Stack : 2b328fa4 2b328fb0 2b328fd8 2b328fdc 2b2f4a6c 00000000 00000d31 00000000
[    5.712000]         00001000 00000000 66040020 616d7269 656e7964 00000000
[    5.712000] Call Trace:
[    5.712000]
[    5.712000]
[    5.712000] Code: 3c100042  afb200c8  8e029030 <8c920010> afbe00d0  02402021  03a0f021  afbf00d4  afb300cc
[    5.712000] block/134: potentially unexpected fatal signal 11.
[    5.712000]
[    5.712000] Cpu 0
[    5.712000] $ 0   : 00000000 1000a400 00000000 00000000
[    5.712000] $ 4   : 00000000 00000000 7f8f1214 7f8f1218
[    5.712000] $ 8   : 00000000 00000fa5 00000001 fffffff8
[    5.712000] $12   : 00000001 ffff0000 00000010 00420000
[    5.712000] $16   : 00420000 00000000 00420000 00407be4
[    5.712000] $20   : 7f8f1214 7f8f1218 0000003a 2b2c9d8c
[    5.712000] $24   : 00418ffc 2b234160
[    5.712000] $28   : 2b2cf2c0 7f8f0fc8 00000000 00403394
[    5.712000] Hi    : 0000002f
[    5.712000] Lo    : 02762762
[    5.712000] epc   : 0040638c 0x40638c
[    5.712000]     Not tainted
[    5.712000] ra    : 00403394 0x403394
[    5.712000] Status: 0000a413    USER EXL IE
[    5.712000] Cause : 10800008
[    5.712000] BadVA : 00000010
[    5.712000] PrId  : 00019300 (MIPS 24Kc)
[    5.716000] urandom-seed: Seed file not found (/etc/urandom.seed)
[    5.752000] firmadyne: sys_reboot[PID: 1 (procd)]: magic1:fee1dead, magic2:28121969, cmd:0
[    5.752000] procd: - early -
[    6.404000] procd: - ubus -
[    6.468000] procd: - init -
Please press Enter to activate this console.
[   10.968000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   10.980000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   11.412000] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   11.432000] ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.432000] 8021q: adding VLAN 0 to HW filter on device eth0
[   11.432000] ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.488000] device eth0.1 entered promiscuous mode
[   11.492000] device eth0 entered promiscuous mode
[   11.516000] br-lan: port 1(eth0.1) entering forwarding state
[   11.516000] br-lan: port 1(eth0.1) entering forwarding state
[   13.520000] br-lan: port 1(eth0.1) entering forwarding state
[   27.308000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   27.308000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   27.364000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   27.920000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   32.956000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131

MAX1800 login: [   37.976000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131

MAX1800 login: [   43.008000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   48.032000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131
[   53.056000] EXT2-fs (sda1): error: ext2_lookup: deleted inode referenced: 131

Login timed out after 60 seconds
Please press Enter to activate this console.
```

If you check the logs after boot, you won't be able to tell exactly which service is causing the crash. Can you give me some helpful information?

dhje0ng, 2022-05-11:
[WEB_v1133n.zip](https://github.com/attify/firmware-analysis-toolkit/files/8667287/WEB_v1133n.zip)

Hi, I'm adding a firmware file for this issue.

extremecoders-re, 2022-05-14:
Log into the router shell and disable the firewall by running `/etc/init.d/firewall stop`. Then you will be able to connect to `192.168.1.254:80` from a web browser.

adi0x90, 2026-09-25:
The address becoming unreachable partway through boot is worth following up. The thread also points to guest firewall behavior, so I'd want to see what changes inside the guest when access stops.

FAT 2's native `qemu-direct` system mode uses mapped host ports, but guest crashes, firewall changes, and service restarts can still break the connection.

Please add the exact firmware version, backend, serial log, and the point at which the mapped endpoint stops responding in v2. That'll help narrow this down to boot, guest networking, or the service.

adi0x90, 2026-09-26:
I downloaded your `WEB_v1133n.zip` attachment and tested the firmware inside it. Native extraction after [PR #108](https://github.com/attify/firmware-analysis-toolkit/pull/108) recovers the rootfs, with all 3,137 regular files and symlink targets matching independent reference extraction. These extraction checks used case-sensitive APFS.

Please retry with [FAT 2.0.1](https://github.com/attify/firmware-analysis-toolkit/releases/tag/v2.0.1) in a fresh FAT project so an old extraction manifest is not reused. Keep the executable and its matching `share` directory together. Use case-sensitive output storage if the image contains filenames that differ only by case.

I'd like to check the point where access disappears after boot. Please share the current serial log, backend, and whether the web interface stays reachable once startup finishes. I'm keeping this open because the extraction result does not yet confirm that the service remains available.