attify/firmware-analysis-toolkit #71: invaild start byte error.

dhje0ng, 2022-05-10:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fat.py", line 172, in <module>
    main()
  File "./fat.py", line 167, in main
    infer_network(arch, image_id, qemu_dir)
  File "./fat.py", line 113, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 418, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fdaaa1052e0>
command: /home/dhjeong/research/wireless-bugbounty/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh
args: ['/home/dhjeong/research/wireless-bugbounty/firmware-analysis-toolkit/firmadyne/scripts/inferNetwork.sh', '1', 'mipseb']
buffer (last 100 chars): b''
before (last 100 chars): b"l)\r\nUnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 18975: invalid start byte\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 33016
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'
```

Error while establishing network connection after firmware extraction. This error seems to be related to the UTF-8 codec.
I am attaching the firmware file below.

[n104e_ml_12_146.zip](https://github.com/attify/firmware-analysis-toolkit/files/8656215/n104e_ml_12_146.zip)

extremecoders-re, 2022-05-10:
Is this the same firmware as in the other issue #70 ?

dhje0ng, 2022-05-11:
No, it's a different firmware. I will add the firmware to the problem in #70.

extremecoders-re, 2022-05-14:
The error is due to utf-8 codec failing to decode `qemu.initial.serial.log`. There are non-printable characters in that file.

Changing the [line](https://github.com/firmadyne/firmadyne/blob/a6d0cbc38715acf806b394d9ade3ad420dfa80aa/scripts/makeNetwork.py#L300) to the following  in file `firmadyne/scripts/makeNetwork.py` will fix the issue.

```python
data = open(infile, errors="ignore").read()
```
This ignores decoding errors.

adi0x90, 2026-09-25:
This failed after extraction, when the network-inference script tried to read the QEMU serial log as UTF-8. The invalid byte stopped the script, leaving `fat.py` waiting for `Interfaces:`.

In FAT 2, native `qemu-direct` system mode uses host-port forwarding and doesn't run that `inferNetwork.sh`/`makeNetwork.py` parsing step.

Closing this since the old log-parsing step has been replaced on that path. If the same firmware still has a boot or network problem in v2, please include the backend and serial log in a new issue.

closed 2026-09-25 by adi0x90