attify/firmware-analysis-toolkit #72: This error occurred when running the firmware!

chinahuhonglue, 2022-09-15:
sudo ./fat.py  ../../DIR-605LBx_FW209UIb02.bin 

                               __           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: DIR-605LBx_FW209UIb02.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "./fat.py", line 172, in <module>
    main()
  File "./fat.py", line 167, in main
    infer_network(arch, image_id, qemu_dir)
  File "./fat.py", line 113, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/home/china/.local/lib/python3.8/site-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/home/china/.local/lib/python3.8/site-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/home/china/.local/lib/python3.8/site-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f9a915d6490>
command: /home/china/文档/firmware-analysis-toolkit-master/firmadyne/scripts/inferNetwork.sh
args: ['/home/china/文档/firmware-analysis-toolkit-master/firmadyne/scripts/inferNetwork.sh', '1', 'mipseb']
buffer (last 100 chars): b''
before (last 100 chars): b"\x96\x87\xe6\xa1\xa3/firmware-analysis-toolkit-master/firmadyne/scratch//1//image.raw': No such file or directory\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 22673
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

chinahuhonglue, 2022-09-15:
firmware ： [DIR-605LBx_FW209UIb02.zip](https://github.com/attify/firmware-analysis-toolkit/files/9571518/DIR-605LBx_FW209UIb02.zip)

Loora1N, 2022-11-21:
maybe you can try `sudo ./fat.py ../../DIR-605LBx_FW209UIb02.bin --qemu 2.5.0`.

ddorando, 2022-12-21:
So the issue here is that pexpect doesn't support cjk characters - aka any other character than latin ones.
Please try again without using any chinese characters in the path.

adi0x90, 2026-09-25:
Before the EOF, the script reports that it can't find the generated `image.raw`. I can't tell from that trace why the file is missing, or whether the characters in the path caused it.

FAT 2's native `qemu-direct` system mode has its own image-preparation and launch steps, without the old `fat.py`/`inferNetwork.sh` setup.

Closing this since that v1 setup has been replaced on this path. If DIR-605L still fails in v2, please include the firmware version, project path, backend, and preparation/serial logs in a new issue.

closed 2026-09-25 by adi0x90