attify/firmware-analysis-toolkit #73: The browser page appears blank

Loora1N, 2022-09-16:
I was trying to emulate the firmware, but the browser page was blank. The fat.py ran successfully, and I can see it by nmap. 
[image]
[image]
The output was repeat `firmadyne: ioctl: 0x4`
```[  283.524000] firmadyne: ioctl: 0x4
[  283.524000] firmadyne: ioctl: 0x4
[  283.528000] firmadyne: ioctl: 0x4
[  283.532000] firmadyne: ioctl: 0x4
[  283.836000] firmadyne: ioctl: 0x6
[  283.836000] firmadyne: ioctl: 0x6
device ioctl:: Operation not supported
Read wlan sta info failed!
Read wlan sta info failed!
[  284.540000] firmadyne: ioctl: 0x4
[  284.540000] firmadyne: ioctl: 0x4
[  284.540000] firmadyne: ioctl: 0x4
[  284.540000] firmadyne: ioctl: 0x4
[  284.840000] firmadyne: ioctl: 0x6
[  284.840000] firmadyne: ioctl: 0x6
device ioctl:: Operation not supported
Read wlan sta info failed!
Read wlan sta info failed!
[  285.544000] firmadyne: ioctl: 0x4
[  285.544000] firmadyne: ioctl: 0x4
[  285.544000] firmadyne: ioctl: 0x4
[  285.544000] firmadyne: ioctl: 0x4
[  285.848000] firmadyne: ioctl: 0x6
[  285.848000] firmadyne: ioctl: 0x6
device ioctl:: Operation not supported
[mld6.c 1117] implement this
Read wlan sta info failed!
Read wlan sta info failed! 
```

Here is the file.
[DIR822A1_FW103WWb03.zip](https://github.com/attify/firmware-analysis-toolkit/files/9580920/DIR822A1_FW103WWb03.zip)

asabhish, 2022-10-13:
I tried to emulate your firmware file (D-Link DIR822) and it works for me correctly. I am getting the web admin page at 192.168.0.1. I have trying to emulate a couple of DD-WRT firmware for TP-Link WR841N since long, but the best I get is to login through command line sometimes. Mostly it is some semaphore issue in a loop! Attaching the screenshot of the login page below:

[image]

extremecoders-re, 2022-10-16:
Just ran the provided firmware [DIR822A1_FW103WWb03.zip](https://github.com/attify/firmware-analysis-toolkit/files/9580920/DIR822A1_FW103WWb03.zip) with fat. It works without issues.

closed 2022-11-02 by Loora1N