attify/firmware-analysis-toolkit #75: setup.sh fails on fresh kali install due to lsb-core package missing on OS

ddorando, 2022-09-24:
Hi, I attempted to run setup.sh today on a kali install I did a few days ago, but it failed due to lsb-core package not existing in the package sources.

I noticed that lsb-core is deprecated, and seems to be removed from the repositories.
I'll test whether it's actually required or not...

Hellworold, 2022-09-29:
> Hi, I attempted to run setup.sh today on a kali install I did a few days ago, but it failed due to lsb-core package not existing in the package sources.
> 
> I noticed that lsb-core is deprecated, and seems to be removed from the repositories. I'll test whether it's actually required or not...

Hi ddorando,

Is is possible to install without lsb-core, iam running kali and for some reason my setup.sh stops with and error

Package lsb-core is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'lsb-core' has no installation candidate
any idea  how i can fix it ?
I have tried this work around but still not able to install-:
https://stackoverflow.com/questions/71191499/lsb-release-has-no-installation-candidate-while-trying-to-install-on-ubuntu-20-0

ddorando, 2022-10-08:
Heya, it is possible to install without lsb-core, however I couldn't get other deps to compile.
Just remove lsb-core from [https://github.com/attify/firmware-analysis-toolkit/blob/master/setup.sh#L5](here).

hem675, 2022-12-17:
> Hi, I attempted to run setup.sh today on a kali install I did a few days ago, but it failed due to lsb-core package not existing in the package sources.
> 
> I noticed that lsb-core is deprecated, and seems to be removed from the repositories. I'll test whether it's actually required or not...

Same problem for me.I use parrot OS. Seems like there are no LSB modules.Please fix this..

ddorando, 2022-12-18:
So, sorry for the wait. lsb-core or the lsb-* dependencies seem to be not used at all. I'll just create a PR that removes it.

adi0x90, 2026-09-25:
Thanks for digging into this. The connection to Binwalk's distribution detection explains why dropping `lsb-core` from the package list wasn't the whole fix.

The FAT 2 installer reads `/etc/os-release` for its system-package setup and doesn't depend on `lsb-core`. It also no longer runs the old Binwalk dependency script.

Closing this since that setup has been replaced. Please use the [v2 installation instructions](https://github.com/attify/firmware-analysis-toolkit/blob/master/INSTALL.md). If you run into another failure on Kali or Parrot, send the distribution version and installer output in a new issue.

closed 2026-09-25 by adi0x90